mylonics/zephyr-ide #1: I can add add project but can't delete it, and can add build but can't delet build

IOT4566, 2024-05-06:
[image]

I can add add project but can't delete it, and can add build but can't delet build!

rijesha, 2024-05-06:
Thanks for the issue. 
It may be that the GUI is not actually updating. 
Could you try "Zephyr IDE:  Load Projects from File" Command?

rijesha, 2024-05-06:
or restarting VS code and seeing if it deletes correctly?

IOT4566, 2024-05-06:
> Thanks for the issue. 谢谢你的问题。 It may be that the GUI is not actually updating.可能是 GUI 实际上没有更新。 Could you try "Zephyr IDE: Load Projects from File" Command?您可以尝试“Zephyr IDE：从文件加载项目”命令吗？

Yes, this issue is caused by the GUI not being updated, as closing vscode and reopening it reveals that it can be deleted. so, you can call the api of update the GUI after handling the deletion to solve this problem, thank you!

IOT4566, 2024-05-06:
> Thanks for the issue.. It may be that the GUI is not actually updating. Could you try "Zephyr IDE: Load Projects from File" Command?

NO, this issue is not caused by the GUI not being updated, beacause i click buttom for delete project or bulid then closing vscode and reopening it, but  it can not be deleted

rijesha, 2024-05-06:
I have pushed an update. The update should be available on the marketplace in 5 minutes or so.

IOT4566, 2024-05-06:
> I have pushed an update. The update should be available on the marketplace in 5 minutes or so.

OK，I try it a few, thanks!

IOT4566, 2024-05-06:
> I have pushed an update. The update should be available on the marketplace in 5 minutes or so.

I have updated and restart vscode, but the issues still  persists！

rijesha, 2024-05-06:
Ok, so the extension is on 1.0.7?
Could you send me .vscode/settings.json file?

IOT4566, 2024-05-06:
> Ok, so the extension is on 1.0.7? Could you send me .vscode/settings.json file?

[image]
[settings.json](https://github.com/mylonics/zephyr-ide/files/15225278/settings.json)

rijesha, 2024-05-06:
Could you put this in your settings.json file and see if the issue is still there?
"zephyr-ide.use-zephyr-ide-json": false,

rijesha, 2024-05-06:
[image]

When you press delete do you see the popup appear asking you if you are sure?

IOT4566, 2024-05-06:
> Could you put this in your settings.json file and see if the issue is still there? "zephyr-ide.use-zephyr-ide-json": false,

[image]

I put "zephyr-ide.use-zephyr-ide-json": false,, but the issue is still there. when i press delete i can not see the popup appear  asking me  if you are sure

IOT4566, 2024-05-06:
> ![image](https://private-user-images.githubusercontent.com/7819200/328292758-c016f0d6-967f-4857-8af1-d90da566eb3d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3L [...]

https://github.com/mylonics/zephyr-ide/assets/169084509/53d380b5-5754-4b70-962f-8a421bea11fe

rijesha, 2024-05-06:
Could you "Zephyr IDE: Clear Project" and try readding projects.
Or start fresh from a new folder.

IOT4566, 2024-05-06:
> Could you "Zephyr IDE: Clear Project" and try readding projects. Or start fresh from a new folder.

https://github.com/mylonics/zephyr-ide/assets/169084509/48ecebb4-f4ed-4592-9bd8-218621db284c

do you mean by this?

rijesha, 2024-05-06:
Yes. It is interesting that the button is not doing anything. 
It seems like none of the buttons on the GUI are working. Can you add a new build from the GUI or did you add it from the command pallet?

rijesha, 2024-05-06:
can you also share a zip of the folder. (you can leave out the external directory though)

IOT4566, 2024-05-06:
> Yes. It is interesting that the button is not doing anything.是的。有趣的是，该按钮没有执行任何操作。 It seems like none of the buttons on the GUI are working. Can you add a new build from the GUI or can you only add them from the command pallet?GUI 上的所有按钮似乎都不起作用。您可以从 GUI 添加新版本，还是只能从命令托盘添加它们？

> Yes. It is interesting that the button is not doing anything. It seems like none of the buttons on the GUI are working. Can you add a new build from the GUI or can you only add them from the command pallet?

I add a new build from gui

IOT4566, 2024-05-06:
> Yes. It is interesting that the button is not doing anything. It seems like none of the buttons on the GUI are working. Can you add a new build from the GUI or can you only add them from the command pallet?

I add a new build from the gui

rijesha, 2024-05-06:
Ok. Could you open the extension host logs and let me know what happens when you try deleting a build?
Ctrl-Shift-p -> Show Logs -> Extension Host

rijesha, 2024-05-06:
Oh your notifications are just off. 
[image]

It should look like this. Or click on the bell icon to see a notification.
[image]

IOT4566, 2024-05-06:
> can you also share a zip of the folder. (you can leave out the external directory though)

ok, but i no how send a zip file

IOT4566, 2024-05-06:
> Oh your notifications are just off. [...]

oh, yes, it's this caused, it's ok now, thank you very much!
Additionally, when adding a project from a template, can we add a fold path that allows users to specify the project

rijesha, 2024-05-06:
> oh, yes, it's this caused, it's ok now, thank you very much! Additionally, when adding a project from a template, can we add a fold path that allows users to specify the project

Do you mean the ability to manually browse to a desired sample project?
Or do you mean renaming the project folder after it is copied?  (Project blinky is added as project blinky2)
Or do you mean putting the project into a subfolder?  (Project blinky is added as project subfolder/blinky)

IOT4566, 2024-05-06:
> Do you mean the ability to manually browse to a desired sample project? Or do you mean renaming the project folder after it is copied? (Project blinky is added as project blinky2) Or do you mean [...]

yes i mean putting the project into a subfolder? (Project blinky is added as project subfolder/blinky), the subfolder can be brow by user and the project folder can be rename also

rijesha, 2024-05-06:
Yea that is something I could do. Please make a new issue for it, and I will get to it when I have a chance. To rename the project, the CMakeLists file needs to be parsed and the project like needs to be updated, so that adds a little bit of complexity.

closed 2024-05-06 by rijesha