mylonics/zephyr-ide #2: Please add project subfolder and rename project name when create a new project from a template

IOT4566, 2024-05-06:
when we create a new project from a template, we holp can putting the project into a subfolder (Project blinky is added as project subfolder/blinky2), the  project subfolder can be brow by user and the project folder(blinky) can be rename blink2 also, thanks!

black-ghost-off, 2024-05-18:
Try it with last version

IOT4566, 2024-05-18:
> Try it with last version

Try it the last version 1.0.22, This feature is not yet supported

rijesha, 2024-05-21:
In v1.0.23 the sample projects are properly renamed.

i.e blinky -> blinky5

If you want to save to a subdirectory you can just put that as a prefix.
i.e blinky -> test/blinky5

You can not browse to a subdirectory with the open dialog. I could not think of a way to browse to a subdirectory without making it inconvenient for users who did not need to use the open dialog.

If you have any suggestions let me know.

closed 2024-05-21 by rijesha