mylonics/zephyr-ide #3: The pop-up information during building does not match the actual situation

IOT4566, 2024-05-06:
[image]
[image]
[image]

The pop-up information during compilation does not match the actual situation.
I add 3 build for blink,. pic1, I am building "nrf52840dk_nrf52840" ,but pop-up show "build esp32c3_devkitm", pic2, i am building "esp32c3_devkitm" ,but pop-up show "build tlsr9518adk80d", pic3, i am building "tlsr9518adk80d" ,but pop-up show "build esp32c3_devkitm"
I found a rule that pop ups display information from the previous compilation, not the current compilation

rijesha, 2024-05-06:
Thank you again for the issue. Ill have a look.

rijesha, 2024-05-06:
I have pushed an update 1.0.9. It should be fixed there.

IOT4566, 2024-05-06:
> I have pushed an update 1.0.9. It should be fixed there.

yes, 1.0.9 is ok, thanks!

closed 2024-05-06 by rijesha