mylonics/zephyr-ide #4: press build pristine button cann't build successful but clean build folder manually

IOT4566, 2024-05-06:
[image]

The blinky_pwm\build folder is the result of a previous build error，but when i press build pristine button，the build folder has not been clean, but instead continues the previous error. At this time, the build folder needs to be manually deleted and then press build pristine button again to successfully build. Therefore, when the user clicks the build pristine button,  zephyr-ide should clean the build folder first and then regenerated to ensure that each compilation is new?

rijesha, 2024-05-06:
The build pristine is a west feature.
West doesn't support moving the project folder like that. So I don't think this should be handled by the zephyr-ide extension.

IOT4566, 2024-05-06:
> The build pristine is a west feature. West doesn't support moving the project folder like that. So I don't think this should be handled by the zephyr-ide extension.

Yes,  you are right .The build pristine is a west feature , but if the build folder is cleared before the build pristine is running, such errors can be handled well and will not have any impact on West

closed 2024-05-15 by IOT4566