mylonics/zephyr-ide #5: Unable to open. dts file for non arm architecture

IOT4566, 2024-05-07:
[image]
[image]
[image]
 
Unable to open. dtc file for non arm architecture。pic1:when i click "Board nrf52840dk_nrf52840" item, then auto to open nrf52840dk_nrf52840.dts, but when i when i click "Board tlsr9518adk80d" item, tlsr9518adk80d.dts file cann't be open. 
Later, I found that it was because the function parameter in pic2  was fixed to ‘arm’, changing this parameter to the arch type of the board can solve this problem。
Also, can we add the case "openBoardDir" in pic2 so that when clicking on "Board Dir...", the item can jump to the corresponding board dir.
Thanks!

rijesha, 2024-05-07:
Thanks for the issue. I will get a fix going.

rijesha, 2024-05-07:
I fixed this in version 1.0.10. Should be available in a few minutes.

IOT4566, 2024-05-07:
> I fixed this in version 1.0.10. Should be available in a few minutes.

https://github.com/mylonics/zephyr-ide/assets/169084509/8c87240f-0540-4701-8d0b-b53adec742b8

cann't open non arm board dts files still

rijesha, 2024-05-07:
It works for me when I do the same board.
Can you see if the log shows anything: Ctrl-Shift-p -> Show Logs -> Extension Host
Can you also send the settings.json file?

IOT4566, 2024-05-07:
> It works for me when I do the same board. Can you see if the log shows anything: Ctrl-Shift-p -> Show Logs -> Extension Host Can you also send the settings.json file?

[image]

IOT4566, 2024-05-07:
> It works for me when I do the same board. Can you see if the log shows anything: Ctrl-Shift-p -> Show Logs -> Extension Host Can you also send the settings.json file?

[image]

arch of esp32c3 is not arm

rijesha, 2024-05-07:
Can you also send the settings.json file?

rijesha, 2024-05-07:
When I did this with a fresh build it works fine.

IOT4566, 2024-05-07:
> When I did this with a fresh build it works fine.

[settings.json](https://github.com/mylonics/zephyr-ide/files/15239022/settings.json)

rijesha, 2024-05-07:
So in the settings file the relBoardSubDir is "relBoardSubDir": "arm\\esp32c3_devkitm".

Did you manually change the setting file? Any build defined before v1.0.10 would need to have the setting files manually changed. 
Could you also make a new build and post what the relBoardSubDir is?

IOT4566, 2024-05-07:
> So in the settings file the relBoardSubDir is "relBoardSubDir": "arm\esp32c3_devkitm".
> 
> Did you manually change the setting file? Any build defined before v1.0.10 would need to have the setting files manually changed. Could you also make a new build and post what the relBoardSubDir is?

no, the settings file is generate auto

IOT4566, 2024-05-07:
The settings file is generate auto before 1.0.10, it not match 1.0.10

rijesha, 2024-05-07:
Ok. So any builds made before v1.0.9 (that are not arm) you will need to manually set the relBoardSubDir.

rijesha, 2024-05-07:
By build I mean build configuration, not the action of performing a "build" or "build pristine".

rijesha, 2024-05-07:
This should fix the problem with the other issue as well.

IOT4566, 2024-05-07:
> This should fix the problem with the other issue as well.这也应该解决另一个问题的问题。

ok

IOT4566, 2024-05-07:
OK,this issuse is fixed

closed 2024-05-07 by rijesha