mylonics/zephyr-ide #6: Add runner to blink project for esp32c3_devkitm board,but runner error

IOT4566, 2024-05-07:
[image]
[image]
[image]
[image]
[image]

 1. pic1. Add runner to blink project for esp32c3_devkitm board, select default runner, but cann't run. 
pic2, I run cmd on powershell,  successful.  when select default runner , -r  (or --runner) is not given?

2. ESP32C3' runner is esp32 and openocd(see pic2, pic3, pic3), but runner 'esp32' is not in the runner list. 

3.  pic5 mean that for a specific board, the runner can be found by reading the board.cmake file under the board folder(see pic4 ), and other runners  that not belong to this board should not be in the runner list when user press add runner button 。

rijesha, 2024-05-07:
Thanks for the issue. I will have a look at parsing the file.

rijesha, 2024-05-07:
I added this into version 1.0.10. Should be available in a few minutes.

rijesha, 2024-05-07:
builds that have already been made will have to be remade or the relBoardSubDir should be added manually into the settings.json file for preexisting builds.

IOT4566, 2024-05-07:
> builds that have already been made will have to be remade or the relBoardSubDir should be added manually into the settings.json file for preexisting builds.

https://github.com/mylonics/zephyr-ide/assets/169084509/7624c808-d741-4ba8-b8e0-b8c6d54012d8

update to 1.0.10, but cann't add runner yet

IOT4566, 2024-05-07:
> builds that have already been made will have to be remade or the relBoardSubDir should be added manually into the settings.json file for preexisting builds.

already rebuild,but cann't add runner

rijesha, 2024-05-07:
It is working for me. Is there anything here?
Ctrl-Shift-p -> Show Logs -> Extension Host

IOT4566, 2024-05-07:
> It is working for me. Is there anything here? Ctrl-Shift-p -> Show Logs -> Extension Host

2024-05-08 01:04:27.424 [error] Error: ENOENT: no such file or directory, open 'e:\zephyr\external\zephyr\boards\arm\esp32c3_devkitm\board.cmake'
	at Object.openSync (node:fs:605:3)
	at Object.func [as openSync] (node:electron/js2c/node_init:2:2214)
	at Object.readFileSync (node:fs:473:35)
	at t.readFileSync (node:electron/js2c/node_init:2:9771)
	at runnerSelector (c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.10\out\project_utilities\runner_selector.js:51:31)
	at addRunnerToBuild (c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.10\out\project_utilities\project.js:479:61)
	at a.value (c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.10\out\panels\project_tree_view\ProjectTreeView.js:486:52)
	at o.y (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:82:660)
	at o.fire (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:82:877)
	at w.$onMessage (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:145:14048)
	at c.S (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:150:5505)
	at c.Q (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:150:5271)
	at c.M (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:150:4361)
	at c.L (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:150:3579)
	at a.value (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:150:2227)
	at o.y (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:82:660)
	at o.fire (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:82:877)
	at u.fire (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:107:14175)
	at a.value (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:176:8023)
	at o.y (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:82:660)
	at o.fire (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:82:877)
	at u.fire (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:107:14175)
	at MessagePortMain.<anonymous> (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:176:6303)
	at MessagePortMain.emit (node:events:517:28)
	at MessagePortMain._internalPort.emit (node:electron/js2c/utility_init:2:2285)
2024-05-08 01:04:34.061 [error] Error: ENOENT: no such file or directory, open 'e:\zephyr\external\zephyr\boards\arm\esp32c3_devkitm\board.cmake'
	at Object.openSync (node:fs:605:3)
	at Object.func [as openSync] (node:electron/js2c/node_init:2:2214)
	at Object.readFileSync (node:fs:473:35)
	at t.readFileSync (node:electron/js2c/node_init:2:9771)
	at runnerSelector (c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.10\out\project_utilities\runner_selector.js:51:31)
	at addRunnerToBuild (c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.10\out\project_utilities\project.js:479:61)
	at a.value (c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.10\out\panels\project_tree_view\ProjectTreeView.js:486:52)
	at o.y (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:82:660)
	at o.fire (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:82:877)
	at w.$onMessage (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:145:14048)
	at c.S (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:150:5505)
	at c.Q (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:150:5271)
	at c.M (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:150:4361)
	at c.L (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:150:3579)
	at a.value (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:150:2227)
	at o.y (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:82:660)
	at o.fire (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:82:877)
	at u.fire (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:107:14175)
	at a.value (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:176:8023)
	at o.y (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:82:660)
	at o.fire (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:82:877)
	at u.fire (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:107:14175)
	at MessagePortMain.<anonymous> (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:176:6303)
	at MessagePortMain.emit (node:events:517:28)
	at MessagePortMain._internalPort.emit (node:electron/js2c/utility_init:2:2285)

IOT4566, 2024-05-07:
> It is working for me. Is there anything here? Ctrl-Shift-p -> Show Logs -> Extension Host

[image]

IOT4566, 2024-05-07:
[image]

Now can add runner for esp32c3_devkitm board, but when select default runner ,the error here still,  in this item, "-r default" in cmd should remove

rijesha, 2024-05-07:
I have pushed an update to remove the "-r default" it is in v1.0.11

IOT4566, 2024-05-07:
OK,this issuse is fixed perfect also

IOT4566, 2024-05-07:
> I have pushed an update to remove the "-r default" it is in v1.0.11

[image]

default runner lose args, should be added

rijesha, 2024-05-07:
Update pushed.

IOT4566, 2024-05-07:
> Update pushed.

OK, this issuse is fixed also

closed 2024-05-08 by rijesha