mylonics/zephyr-ide #10: how to start menuconfig on zephyr-ide GUI

IOT4566, 2024-05-08:
[image]

I can run 'west build -t menuconfig' cmd in powershell, but i cann't find entry button on zephyr-ide gui

rijesha, 2024-05-08:
Im not sure where the best place for this to go in the GUI would be. Im thinking maybe there should be another panel showing the configuration. 
Or do you think this should just be in the command pallete as opposed to a GUI button?

IOT4566, 2024-05-08:
> Im not sure where the best place for this to go in the GUI would be. Im thinking maybe there should be another panel showing the configuration. Or do you think this should just be in the command pallete as opposed to a GUI button?

I thinking the kconfig and devicetree be another panel showing the configuration best.

IOT4566, 2024-05-08:
I mean the menuconfig and devicetree config  be another panel showing the configuration best. we can refer nrf connect extend.

black-ghost-off, 2024-05-17:
west build -t menuconfig -b %current selected board%

add that to 
[image]

rijesha, 2024-05-21:
Added in 1.0.23 in the active build panel

closed 2024-05-21 by rijesha