mylonics/zephyr-ide #9: Can you add the selection of Zephyr version on the GUI

IOT4566, 2024-05-08:
Can you add the selection of Zephyr version on the GUI，For example, the current default installation of Zephyr is the latest version 3.6, but I want to use LTS version 2.7.6

rijesha, 2024-05-08:
I dont really know the best way of implementing this. I could add a 2.7.6 template and call west init again after the user has deleted the .west folder.

IOT4566, 2024-05-08:
> I dont really know the best way of implementing this. I could add a 2.7.6 template and call west init again after the user has deleted the .west folder.

Allow users to choose the desired zephyr version during the west init phase. and add a button to switch the Zephyr version after the Reset extension button？

IOT4566, 2024-05-08:
You can also refer to the ESP-IDF extension

rijesha, 2024-05-13:
I added the ability to select a version during west init. So if you want to change the version just run west init and west update again. The change is in v1.0.14.

IOT4566, 2024-05-14:
https://github.com/mylonics/zephyr-ide/assets/169084509/5b1e6e86-03c7-43c5-85c3-d9856ef46a2b

2024-05-14 19:16:00.592 [error] Error: Cannot find module 'js-yaml'
Require stack:
- c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\setup_utilities\west_selector.js
- c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\setup_utilities\setup.js
- c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\utilities\utils.js
- c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\extension.js
- d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js
- d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\bootstrap-amd.js
- d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\bootstrap-fork.js
	at Module._resolveFilename (node:internal/modules/cjs/loader:1084:15)
	at i._resolveFilename (node:electron/js2c/utility_init:2:13405)
	at Module._load (node:internal/modules/cjs/loader:929:27)
	at c._load (node:electron/js2c/node_init:2:13672)
	at v._load (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:176:5607)
	at i._load (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:173:29719)
	at t._load (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:138:35317)
	at Module.require (node:internal/modules/cjs/loader:1150:19)
	at Module.require (c:\Users\yangfan\.vscode\extensions\nordic-semiconductor.nrf-devicetree-2024.3.78\dist\extension.js:27:42010)
	at h (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js:4:647)
	at pickWestYml (c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\setup_utilities\west_selector.js:127:26)
	at async MultiStepInput.stepThrough (c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\utilities\multistepQuickPick.js:26:24)
	at async collectInputs (c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\setup_utilities\west_selector.js:209:9)
	at async westSelector (c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\setup_utilities\west_selector.js:212:19)
	at async westInit (c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\setup_utilities\setup.js:529:25)
	at async c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\extension.js:119:13
	at async r.h (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:153:189457) zephyr-ide.west-init {"value":"mylonics.zephyr-ide","_lower":"mylonics.zephyr-ide"}
2024-05-14 19:16:01.599 [error] Error: Cannot find module 'js-yaml'
Require stack:
- c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\setup_utilities\west_selector.js
- c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\setup_utilities\setup.js
- c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\utilities\utils.js
- c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\extension.js
- d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js
- d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\bootstrap-amd.js
- d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\bootstrap-fork.js
	at Module._resolveFilename (node:internal/modules/cjs/loader:1084:15)
	at i._resolveFilename (node:electron/js2c/utility_init:2:13405)
	at Module._load (node:internal/modules/cjs/loader:929:27)
	at c._load (node:electron/js2c/node_init:2:13672)
	at v._load (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:176:5607)
	at i._load (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:173:29719)
	at t._load (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:138:35317)
	at Module.require (node:internal/modules/cjs/loader:1150:19)
	at Module.require (c:\Users\yangfan\.vscode\extensions\nordic-semiconductor.nrf-devicetree-2024.3.78\dist\extension.js:27:42010)
	at h (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js:4:647)
	at pickWestYml (c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\setup_utilities\west_selector.js:127:26)
	at async MultiStepInput.stepThrough (c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\utilities\multistepQuickPick.js:26:24)
	at async collectInputs (c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\setup_utilities\west_selector.js:209:9)
	at async westSelector (c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\setup_utilities\west_selector.js:212:19)
	at async westInit (c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\setup_utilities\setup.js:529:25)
	at async c:\Users\yangfan\.vscode\extensions\mylonics.zephyr-ide-1.0.15\out\extension.js:119:13
	at async r.h (d:\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:153:189457)

rijesha, 2024-05-14:
I forgot to package it properly. I have issued a fix on 1.0.17.

IOT4566, 2024-05-14:
> I forgot to package it properly. I have issued a fix on 1.0.17.

> I forgot to package it properly. I have issued a fix on 1.0.17.

[image]
[image]
error here still

rijesha, 2024-05-14:
Ok. I think I have added it correctly now in 1.0.18

IOT4566, 2024-05-14:
> Ok. I think I have added it correctly now in 1.0.18. It should be up in a few minutes.

[image]
[image]
1.0.18，can start,but error here still

rijesha, 2024-05-14:
Thanks for the issue and testing the code. I will close this issue. If you would like this new feature please make a new issue and include a list of what would be user selectable. Also feel free to make PRs as well.

closed 2024-05-14 by rijesha

IOT4566, 2024-05-14:
[image]

click west init , select nrf connect config, the error here

rijesha, 2024-05-14:
I pushed an update just now to remove the 
```javascript
self: 
    path:
``` 
completely