mylonics/zephyr-ide #12: Custom flash loader for "teensy"

black-ghost-off, 2024-05-13:
https://github.com/mylonics/zephyr-ide/issues/11

The build is going well. However, the problem now is that I can't add a runner for flash teensy (there is a custom bootloader without swd/jtag and it is necessary only through their utility ( https://www.pjrc.com/teensy/loader_cli.html )

black-ghost-off, 2024-05-13:
I know. However, zephyr natively supports teensy_cli (you need to specify the path of the utility), I mean it should work in our case out of the box)

Offtop: I have already tested xiao seeed ble(nrf52840), nucleo h563zi, nrf52840 devkit, everything works fine with the IDE
Now in progress testing xiao seeed ESP32s3, esp32s3 devkit n8r8

rijesha, 2024-05-13:
could you give me the west command you would run to load the code.

black-ghost-off, 2024-05-13:
Just 
> west flash 
[image]

black-ghost-off, 2024-05-13:
path "teensy_loader_cli.exe" in added to PATH

rijesha, 2024-05-13:
I pushed an update (1.0.15) that allows the default west flash to be run for this board. You will have to pass the runner arguments when you setup the runner. ("-H -r teensy")

black-ghost-off, 2024-05-13:
[image]
[image]

If built with https://docs.zephyrproject.org/latest/develop/getting_started/index.html
But the project that is created by the IDE does not lend itself to 
>west flash --build-dir “.\build\teensy40\”

rijesha, 2024-05-13:
What is the build command that you use?

 The error is correct and the teensy folder does not have a board.cmake file I think that is why there is no zephyr_runners_yaml

rijesha, 2024-05-13:
What version of zephyr did you use for the build that works and what version are you using with zephyr-ide?

rijesha, 2024-05-13:
Ok. I think the problem is that you are using the v3.6.0 release of zephyr. That version doesn't have support for the teensy cli flasher.

Change the revision in application/west.yml file from v3.6.0 to main and then run a west update.

rijesha, 2024-05-13:
You will also need to remake the build configurations, because main has the board directories reorganized.

IOT4566, 2024-05-14:
Zephyr ide retrieves the supported runners for this board by traversing board.cmake file in the board directory, but unfortunately, teensy4 does not include board.cmake, while other boards do

IOT4566, 2024-05-14:
Zephyrproject requirements the`board.cmake` file must have

black-ghost-off, 2024-05-14:
I change 3.6.0 to main, and that fix problem well. For feature add option for select version of zephyr?

closed 2024-05-14 by black-ghost-off

rijesha, 2024-05-14:
The west init function should now have that ability.