mylonics/zephyr-ide #14: Bug: Two versions of one board

black-ghost-off, 2024-05-14:
[image]
[image]

When selecting the ESP32s3 devkit board (in my case esp32s3 devkit c(n8r8), I see two choices, esp32s3_devkitc_appcpu and esp32s3_devkitc_procpu, although only esp32s3_devkitc is in the list of available boards

black-ghost-off, 2024-05-14:
Versions:
west: v1.2.0
zephyr: main
SDK: 0.16.2

(Off-topic, add a feature somewhere so that you can get all versions of all packages)

rijesha, 2024-05-14:
Where is the 2nd screenshot from? It sounds like that board has two cpu cores on it and you need to select what core you are building for.
I would imagine one of them is the bluetooth core.

rijesha, 2024-05-14:
> Versions: west: v1.2.0 zephyr: main SDK: 0.16.2
> 
> (Off-topic, add a feature somewhere so that you can get all versions of all packages)

What do you mean all versions of all packages?

black-ghost-off, 2024-05-14:
2nd screenshot it`s if there is an error that such a board does not exist, ``west build`` displays a list of all supported boards

black-ghost-off, 2024-05-14:
> What do you mean all versions of all packages?

One button for copy version of sdk, west, zephyr, arm packages, esp pacakges, etc...

rijesha, 2024-05-14:
This issue is closed. Select the correct processor for the board you are building and you should be good.

black-ghost-off, 2024-05-14:
This is an issue that there is no correct configuration in the build configuration selection

black-ghost-off, 2024-05-14:
@rijesha reopen issue

rijesha, 2024-05-14:
I dont think I understand the issue. What are you expecting in the build configuration selection.

black-ghost-off, 2024-05-14:
The list when selecting the build configuration contains those boards that do not exist, and the one that is needed is not there. You only need esp32s3_devkitc WITHOUT appcpu, procpu

rijesha, 2024-05-15:
Also if you are having build issues with that board, this may be relevant.
https://github.com/zephyrproject-rtos/zephyr/issues/72639

black-ghost-off, 2024-05-15:
> Also if you are having build issues with that board, this may be relevant.
> https://github.com/zephyrproject-rtos/zephyr/issues/72639

[image]

black-ghost-off, 2024-05-15:
Doesn't meter, main always on 0 core boot(only if you wasn't touch idf menuconfig)

Two cores controlled by rtos, that cores doesn't have two different firmwares

We need select JUST "esp32s3_devkitc" for build firmware 

I don't know where from appcpu and *second name", it's not in list of available board(esp32s3_devkitc_appcpu and esp32_divkitc_otherboard)

My flow
Create new project
Select build configuration 
Select BOARD esp32s3_devkitc
BUT configuration menu give me list without "ZEPHYR SUPPORTED BOARD", it's give me "(esp32s3_devkitc_appcpu and esp32_divkitc_otherboard)"

THERE BUG, WHAT WE NEED ESP32S3_DEVKITC(WITHOUT ANYTHING ELSE)

black-ghost-off, 2024-05-15:
Bug in list of your zephyr IDE, build configuration 
List where you select which one board you want to use

rijesha, 2024-05-15:
Please post the west build command that you would use if you weren't using zephyr ide.

black-ghost-off, 2024-05-15:
https://github.com/mylonics/zephyr-ide/assets/70693212/47fbd7a2-b33e-4d1f-8fd6-e794d32bb6f9

black-ghost-off, 2024-05-15:
https://docs.zephyrproject.org/latest/boards/espressif/esp32s3_devkitc/doc/index.html

black-ghost-off, 2024-05-15:
I suspect that the bug may be in the SDK/West/Zephyr itself

rijesha, 2024-05-15:
There is a bug with the zephyr ide app. Hopefully I will be able to have a look at it today.

black-ghost-off, 2024-05-15:
One observation. 
>west build -besp32s3_devkitc 
is also not possible, it is necessary as I dropped in the link earlier

rijesha, 2024-05-15:
So there have been a lot of changes recently with the way the boards are being handled.

I think the zephyr ide works correctly if you are using v3.6.0. 
I made some changes that work with main, but that will break v3.6.0 and lower.

I would recommend using v3.6.0 for the time being. If using v3.6.0 then you can use esp32s3_devkitc as you pointed out earlier. The zephyr documentation and zephyr main have all changed to using esp32s3_devkitc/esp32/appcpu.

If you would really like to use main, you can manually edit the board for the project in settings.json.

rijesha, 2024-05-15:
I got my changes to work for both the new and older version of zephyr. It is in v1.20.0

black-ghost-off, 2024-05-16:
LGMT, It works

closed 2024-05-16 by black-ghost-off