mylonics/zephyr-ide #16: please add custom west.yml item on west init

IOT4566, 2024-05-14:
I think it would be better to handle the west init : first, let the user choose the version number, such as 3.6 or 2.7.6 (LTS) or other version, next there should be an option to list all the supported architectures for that version in the western yml step, and provide a checkbox for users to select multiple. Finally, generate the western yml file based on the user's selection and initialize it
for example, user select nordic, espressif and telink then the west.yml will include flowing module
.- hal_noridic
.- hal_espressif
.- hal_telink

rijesha, 2024-05-14:
What should the full list of items look like? And in your opinion should other modules be included?

IOT4566, 2024-05-15:
> What should the full list of items look like? And in your opinion should other modules be included(i.e libmetal, lvgl)?

[image]
The the full list of items is the zephyrproject supported all hal_xxx l

rijesha, 2024-05-21:
Added in 1.0.23

closed 2024-05-21 by rijesha