mylonics/zephyr-ide #17: 1.0.20 broke board selection

rpiper, 2024-05-16:
After upgrading to 1.0.20, I'm getting this error about ZEPHYR_BASE not being defined. 
[image]

I added the Environment Variable ZEPHYR_BASE with the path the zephyr directory. I didn't need this environment variable before 1.0.20, but instead used the Kconfig extension to set the zephyr base on a per-repository basis in the workspace settings like this:
`"kconfig.zephyr.base": "c:\workspace\zephyr",`

After ZEPHYR_BASE was added to Environment Variables, not I get a different python error.
[image]

This same board was working prior to 1.0.20, so I think it is the new board_list.py that is causing the issues.

rijesha, 2024-05-16:
I think I have it fixed in v1.0.21.

I am not sure about the environment variable issue. If you call the west update from the GUI/command palette, that should set the ZEPHYR_BASE variable.

I included a new command to try to debug the issue. Can you run ```Zephyr IDE: Debug Internal Shell``` and see if there is a ZEPHYR_BASE variable present?

rpiper, 2024-05-16:
> Thanks for the issue. What version of zephyr are you using?

I am using Zephyr 3.5.0

> I think I have it fixed in v1.0.21.
>
> I am not sure about the environment variable issue. If you call the west update from the GUI/command palette, that should set the ZEPHYR_BASE variable.
>
> I [...]

Thanks, v1.0.21 fixed the board selection, and it works now even without me manually defining ZEPHYR_BASE.

How do I go about running `Zephyr IDE: Debug Internal Shell` to see if there is a ZEPHYR_BASE variable present? Is there a button in the extension UI, or is there a command I have to enter?

rijesha, 2024-05-16:
open the command pallet and run it from there.
ctrl-p -> Zephyr IDE: Debug Internal Shell

rpiper, 2024-05-16:
I do not see that option in v1.0.21
[image]

rpiper, 2024-05-16:
`ALLUSERSPROFILE=C:\ProgramData
APPDATA=C:\Users\rpiper\AppData\Roaming
ChocolateyInstall=C:\ProgramData\chocolatey
ChocolateyLastPathUpdate=133543874573455647
CHROME_CRASHPAD_PIPE_NAME=\\.\pipe\crashpad_21952_CFIQBYLDGOZORVYI
CLIENTNAME=FRACTALII
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPUTERNAME=AXILLA
ComSpec=C:\Windows\system32\cmd.exe
DriverData=C:\Windows\System32\Drivers\DriverData
ELECTRON_RUN_AS_NODE=1
GOPATH=C:\Users\rpiper\go
HOMEDRIVE=C:
HOMEPATH=\Users\rpiper
IVIROOTDIR32=C:\Program Files (x86)\IVI Foundation\IVI\
IVIROOTDIR64=C:\Program Files\IVI Foundation\IVI\
KMP_DUPLICATE_LIB_OK=TRUE
LOCALAPPDATA=C:\Users\rpiper\AppData\Local
LOGONSERVER=\\AXILLA
MKL_SERIAL=YES
NUMBER_OF_PROCESSORS=40
OneDrive=C:\Users\rpiper\OneDrive - TASI
OneDriveCommercial=C:\Users\rpiper\OneDrive - TASI
ORIGINAL_XDG_CURRENT_DESKTOP=undefined
OS=Windows_NT
Path=c:\patriot\patriot\.venv\Scripts;\;c:\patriot\patriot\.venv\bin;\;c:\patriot\patriot\.venv\Scripts;\;c:\patriot\patriot\.venv\bin;\;c:\patriot\patriot\.venv\Scripts;\;c:\patriot\patriot\.venv\bin;\;c:\patriot\patriot\.venv\Scripts;\;c:\patriot\patriot\.venv\bin;\;c:\patriot\patriot\.venv\Scripts;\;c:\patriot\patriot\.venv\bin;\;c:\patriot\patriot\.venv\Scripts;\;c:\patriot\patriot\.venv\bin;\;c:\patriot\patriot\.venv\Scripts;\;c:\patriot\patriot\.venv\bin;\;c:\patriot\patriot\.venv\Scripts;\;c:\patriot\patriot\.venv\bin;\;C:\Python312\Scripts\;C:\Python312\;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Oracle\Java\java8path;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\PuTTY\;C:\Program Files\Microsoft VS Code\bin;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin\;C:\ProgramData\chocolatey\bin;C:\Program Files\7-Zip;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Program Files\otii;C:\Program Files\Keysight\IO Libraries Suite\bin;C:\Program Files\IVI Foundation\VISA\Win64\ktvisa;C:\Program Files (x86)\Keysight\IO Libraries Suite\bin;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\ktvisa;C:\Program Files (x86)\IVI Foundation\IVI\Bin\;C:\Program Files\IVI Foundation\IVI\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\winnt\agvisa;C:\Program Files\SafeNet\Authentication\SAC\x64;C:\Program Files\SafeNet\Authentication\SAC\x32;C:\Program Files (x86)\Microchip\MPLAB C32 Suite\bin;C:\Program Files\Go\bin;C:\Program Files\dotnet\;C:\Program Files\TortoiseGit\bin;C:\Program Files\Git\cmd;C:\Program Files\CMake\bin;C:\Program Files\SEGGER\JLink;C:\gnuarmemb\bin;C:\Users\rpiper\AppData\Local\Microsoft\WindowsApps;C:\Users\rpiper\go\bin;C:\Users\rpiper\.dotnet\tools
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC;.PY;.PYW
PROCESSOR_ARCHITECTURE=AMD64
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 143 Stepping 8, GenuineIntel
PROCESSOR_LEVEL=6
PROCESSOR_REVISION=8f08
ProgramData=C:\ProgramData
ProgramFiles=C:\Program Files
ProgramFiles(x86)=C:\Program Files (x86)
ProgramW6432=C:\Program Files
PROMPT=$P$G
PSModulePath=C:\Program Files\WindowsPowerShell\Modules;C:\Windows\system32\WindowsPowerShell\v1.0\Modules
PUBLIC=C:\Users\Public
SESSIONNAME=RDP-Tcp#1
SIGNTOOL=C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\x86\signtool.exe
SystemDrive=C:
SystemRoot=C:\Windows
TEMP=C:\Users\rpiper\AppData\Local\Temp
TMP=C:\Users\rpiper\AppData\Local\Temp
USERDOMAIN=AXILLA
USERDOMAIN_ROAMINGPROFILE=AXILLA
USERNAME=rpiper
USERPROFILE=C:\Users\rpiper
VIRTUAL_ENV=c:\patriot\patriot\.venv
VSCODE_AMD_ENTRYPOINT=vs/workbench/api/node/extensionHostProcess
VSCODE_CODE_CACHE_PATH=C:\Users\rpiper\AppData\Roaming\Code\CachedData\dc96b837cf6bb4af9cd736aa3af08cf8279f7685
VSCODE_CRASH_REPORTER_PROCESS_TYPE=extensionHost
VSCODE_CWD=C:\Program Files\Microsoft VS Code
VSCODE_HANDLES_UNCAUGHT_ERRORS=true
VSCODE_IPC_HOOK=\\.\pipe\900c8b8f-1.89.1-main-sock
VSCODE_L10N_BUNDLE_LOCATION=
VSCODE_NLS_CONFIG={"locale":"en-us","osLocale":"en-us","availableLanguages":{},"_languagePackSupport":true}
VSCODE_PID=21952
VXIPNPPATH=C:\Program Files (x86)\IVI Foundation\VISA\
VXIPNPPATH64=C:\Program Files\IVI Foundation\VISA\
windir=C:\Windows
ZEPHYR_BASE=C:\patriot\zephyr
ZEPHYR_SDK_INSTALL_DIR=C:\Users\rpiper\.zephyr_ide\toolchains`

rpiper, 2024-05-16:
Looks good to me

`ZEPHYR_BASE=C:\patriot\zephyr`
`ZEPHYR_SDK_INSTALL_DIR=C:\Users\rpiper\.zephyr_ide\toolchains`

rpiper, 2024-05-16:
Fixed in v1.0.21

closed 2024-05-16 by rpiper