mylonics/zephyr-ide #18: West update failing when using a Zephyr workspace application

rpiper, 2024-05-17:
I have a zephyr application that follows the [Zephyr workspace application](https://docs.zephyrproject.org/latest/develop/application/index.html#zephyr-workspace-application) hierarchy.

Initializing the workspace and setting up the West environment both work without any problem.
The `west init` and `west update` commands also run without any error. However, the `west list -f {path:28} zephyr` command is returning the error:
```
FATAL ERROR: unknown project name/path: zephyr
  Hint: use "west list" to list all projects.
```

I believe this is due to the Zephyr workspace hierarchy and the way I am selecting west.yml in the workspace.

The workspace looks like this
```
workspace/
├─── .west/
│    └── config
└─── app/
     └── .venv/
     └── west.yml (imports zephyr)
├─── bootloader/
├─── modules/
├─── tools/
├─── zephyr/
│    └── west.yml
```
and I am initializing the workspace by using 'Select west.yml in Workspace' and selecting `workspace/app/west.yml`

I believe the problem is sets `rootPath` to `workspace/app` and runs all the west command with that as the working directory.

Normally, I would run `west init -l app` and `west update` from the top level `workspace/` directory, however both commands work if run from within the `workspace/app` directory (and using the absolute path to west.yml as done in `west init -l ${westSelection.path}`)

For some strange reason, `west list -f {path:28} zephyr` only works if run from the top level `workspace/` directory, and returns an error if run from within the `workspace/app` directory.

Would it be possible to detect this workspace hierarchy and run the `west` commands from the top level `workspace/` directory?

I'm happy to help test any possible workarounds to this issue.

rijesha, 2024-05-17:
Thanks for the issue. 

Yes this workspace hierarchy should work/be supported.

Could you also post the contents of .west/config and the west.yml?

rijesha, 2024-05-17:
So it seems to be working for me. 
I made a new folder then added app/west.yml. Then I pressed initialize workspace and selected app/west.yml.

This is what my workspace looks like.
[image]

I am able to run west update just fine from both the GUI and the Zephyr IDE terminal

rijesha, 2024-05-17:
Is your python environment folder in workspace/.venv

rpiper, 2024-05-17:
> Is your python environment folder in workspace/.venv

No, it is created in workspace/app/.venv because that is where I selected the west.yml file from.

rijesha, 2024-05-17:
Could you post the PATH,VIRTUAL_ENV, and ZEPHYR_BASE when you run the Zephyr IDE: Debug Internal Shell

rijesha, 2024-05-17:
The .venv should always be in the workspace folder. 

It seems like there is some mixup between the root folder when you setup the app. 
When dealing with west you should have your workspace_root and the west.yml must always be in a subdirectory of workspace_root. Your west init must be called from workspace_root.

In vscode when you have opened a folder/workspace, have you opened workspace_root or app?

rpiper, 2024-05-17:
> It seems like there is some mixup between the root folder when you setup the app.
> When dealing with west you should have your workspace_root and the west.yml must always be in a subdirectory of workspace_root. Your west init must be called from workspace_root.

Understood, this is what I suspected was the issue. My workspace_root has subdirectories app and zephyr, and both directories have a west.yml file. The app/west.yml imports the zephyr/west.yml. This is how Nordic SDK apps are organized, so I was thinking I could do the same with Zephyr IDE. I will reorganize my project to comply with this folder structure.

rijesha, 2024-05-17:
I think what you had should have been ok. Isn't it similar to the screenshot I posted above?

Could you share a git repo of your desired workspace that isn't working?

rpiper, 2024-05-17:
> I think what you had should have been ok. Isn't it similar to the screenshot I posted above?

Not quite, I have a separate west.yml in my app directory that clones the zephyr project and then imports the zephy/west.yml. This strategy allows for the app to be in its own directory, separate from zephyr, tools, bootloader, etc.

> Could you share a git repo of your desired workspace that isn't working?

I have setup this example repository for you to test out https://github.com/rpiper/ZephyrWorkspaceApp

rpiper, 2024-05-18:
I found the source of all my problems. I wasn't setting the `workspace.workspaceFolders` correctly in my workspace file.

After finding the bit of code below, I have have adjusted my app.code-workspace file to use the top-level `workspace/` directory as the first folder in the workspace folders array.

```
export function getRootPath() {
  let rootPaths = workspace.workspaceFolders;
  if (rootPaths === undefined) {
    return;
  } else {
    return rootPaths[0].uri;
  }
}
```

The beginning of my app.code-workspace file (located in workspace/app/) now contains this path to make the entire workspace/ directory the workspace.

```
{
	"folders": [
		{
			"path": ".."
		}
	],
```

After this change, the extension is working perfectly! Thanks for all your help, and this amazingly powerful extension!

closed 2024-05-18 by rpiper