mylonics/zephyr-ide #19: Methos to add extra west build and CMake arguments

rpiper, 2024-05-17:
Would it be possible to add some new `"buildConfigs"` options that allow passing optional extra arguments both to `west build` and extra CMake arguments off the end, i.e. `west build ... --  -DEXTRA_ARG=1` ?

I understand these are not commonly used, so they could probably be available to edit in the config.json only, and not the UI.

The use case for the extra argument to `west build` would be to add the `--sysbuild` argument if using [sysbuild to build a multi-image build](https://docs.zephyrproject.org/latest/build/sysbuild/index.html), which is often used when building with the MCUboot bootloader.

There are many possible extra CMake arguments, but a common one that might be used for a release build would be to set
`-DNDEBUG=1`

as in

`west build ... -- -DNDEBUG=1`

I'm happy to help test any possible solution that allows adding these extra arguments to the buildConfig.

rijesha, 2024-05-17:
Yes. This should be doable. Ill put it on the todo list.

hongshui3000, 2024-05-18:
It would be more convenient if CMakePresets.json could be supported like nrf connect for vscode.
[image]

PR #22 (2024-05-20): BuildConfig: Add westBuildArgs and westBuildCMakeArgs
These optional configuration items are used as extra argument strings used in creating the 'west build' command string 'west build ... {westBuildArgs} -- ... {westBuildCMakeArgs}'

Resolves #19
closed 2024-05-20 by rijesha