mylonics/zephyr-ide #25: Add GUI elements for build flags

rijesha, 2024-05-21:
Add GUI elements for westBuildArgs and westBuildCMakeArgs.
Needs to be added to tree view and also may need to be part of build configuration. 
Need to combile westBuildCMakeArgs with build optimization.

rijesha, 2024-05-27:
There is a prerelease version of the app that will be available v1.0.28. That includes the GUI changes to modify the build arguments. @rpiper @hongshui3000 could you have a test out that version and let me know if you have any comments?

Please note that the new version no longer uses the debugOptimization variable and that by using the new version you will not be able to downgrade. (So save a copy of your projects in settings.json (or zephyr-ide.json)

rpiper, 2024-05-28:
The new feature is working great! Opening my existing project shows the Args I had manually added, and when I click on them, they are available to edit in the top bar.

[image]

I also tested clearing all projects and resetting the extension and initializing the workspace, SDK, west, and then adding the project from new, and both Args options were presented after selecting the board, and I saw them added to the new project in my workspace file, and used properly in `west build`.

When I added the new project, I was asked to select an optimization, and after I selected the "use Kconfig" option as I usually do, I do still have this line in my project's workspace file.
`"debugOptimization": "Don't set. Will be configured in included KConfig file",`

I'm not sure if that is still the intended behavior with the new version that doesn't use debugOptimization.

The West Args and CMake Args GUI elements look good to me!

rijesha, 2024-05-29:
Thanks for testing @rpiper. I will remove that string from being generated.

closed 2024-05-29 by rijesha