mylonics/zephyr-ide #26: Please add a menuconfig button for each board.

IOT4566, 2024-05-21:
[image]

Although the active project already includes the menuconfig button, it is obvious that adding the menuconfig button to each board in the project also be convenient

rijesha, 2024-05-27:
There is a prerelease version of the app that will be available v1.0.28. This version adds the button where you indicated.

Please note that the new version no longer uses the debugOptimization variable and that by using the new version you will not be able to downgrade. (So save a copy of your projects in settings.json (or zephyr-ide.json)

IOT4566, 2024-05-28:
thinks！

closed 2024-05-28 by IOT4566