mylonics/zephyr-ide #28: Init workspace from git repoisotry

beriberikix, 2024-05-25:
Love where this extension is going!

It would be great to figure out a flow that sets up a workspace from a git repo. It's very common for projects to use `west init -m {{repo}}` to pull in all the dependencies and setup the folder structure correctly. The official [`example-application`](https://github.com/zephyrproject-rtos/example-application) or one of our [`reference designs`](https://github.com/golioth/reference-design-template?tab=readme-ov-file#use-west-to-initialize-and-install) are typical.

Is that something you'd consider?

rijesha, 2024-05-25:
Hi. Thanks for the suggestion.

I have implemented the ability to pass a git repo during the West Init/Initialize Workspace command (v1.0.25). Give it a try a let me know if this is what you are looking for.

beriberikix, 2024-05-26:
Yes! Thank you for the fast turnaround :raised_hands:

closed 2024-05-26 by beriberikix