mylonics/zephyr-ide #31: Toolchain fails to install on 1.0.25

beriberikix, 2024-05-26:
[image]

The Install SDK fails as shown above. I've tried different SDK versions, same result. It was working on 1.0.24 I'm running Ubuntu 23.10 but let me know if there's anything else I can provide to help troubleshoot!

rijesha, 2024-05-26:
Thanks for the issue. I'll have a look at it later today in ubuntu. It seems to be still working for me in windows.

rijesha, 2024-05-26:
I'm not quite sure what the problem is. It looks like the packaging of node modules in that version somehow failed. I have issued a new update 1.0.27 that seems to work during my testing. (But I may have some preinstalled node modules that is allowing it work on my computer. Not sure). The new version does seem to work on a VM for me.

Could you try out the new version when you have time and let me know if the problem goes away.

beriberikix, 2024-05-26:
That fixed it!

closed 2024-05-26 by beriberikix