mylonics/zephyr-ide #33: Detect Project/Examples in modules

beriberikix, 2024-05-27:
The extension has a neat feature in that it automatically scans `samples` directory of Zephyr and displays them as template to use as a starting point for Projects. External modules also include samples but the sample doesn't currently scan for those. It would be great to include in the list for the "Create a Project from Template" command.

I'm not sure if there's a convention across module authors, but I'd assume `sample` and `example` are common.

hongshui3000, 2024-05-27:
Have a scan filter or be able to specify scan directories. I think the user experience will be better. If there are too many modules or examples are added later in zephyr, scanning them all will take a lot of time, and it will be difficult to find the examples you need.

rijesha, 2025-04-14:
So the zephyr way of doing this is to specify a sample directory in your module.yml file

```
samples:
  - samples
```

The extension (prerelease 1.6.8) now looks for samples in the directory specified by module.yml and lists them.

rijesha, 2025-04-20:
In release v1.7.0

closed 2025-04-20 by rijesha