mylonics/zephyr-ide #34: Issue creating build for Golioth SDK

beriberikix, 2024-05-27:
This may be specific to our SDK but it seems like the extension is having issues finding `boards` when deeply nested in a module.

Steps to reproduce:

- Initialize the workspace via git (`https://github.com/golioth/golioth-firmware-sdk`, `--mr v0.12.2 --mf west-zephyr.yml`)
- Import the `hello` project at `modules/lib/golioth-firmware-sdk/examples/zephyr/hello`
- Select `add build` and navigate to `modules/lib/golioth-firmware-sdk/examples/zephyr/hello/boards`

The build will fail to add and you should see the following output from the extension:

```
Traceback (most recent call last):
  File "/home/jberi/.vscode/extensions/mylonics.zephyr-ide-1.0.27/scripts/get_board_list.py", line 51, in <module>
    get_boards(args)
  File "/home/jberi/.vscode/extensions/mylonics.zephyr-ide-1.0.27/scripts/get_board_list.py", line 33, in get_boards
    for board in list_boards.find_v2_boards(args):
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'list_boards' has no attribute 'find_v2_boards'. Did you mean: 'find_boards'?
```

rpiper, 2024-05-27:
The `modules/lib/golioth-firmware-sdk/examples/zephyr/hello/boards` directory looks like it only contains Kconfig .conf and devicetree .overlay files, and not any actual custom board definitions.

I think you have to first select the board from the `zephyr/boards/...` directory, and then the these overlays in the `modules/lib/golioth-firmware-sdk/examples/zephyr/hello/boards` directory should get included automatically when generating the Kconfig and DTC files.

For instance, if I build the `hello` project for the `nrf9160dk_nrf9160_ns`, I would select the board like this:
[image]

and then when I build, I can see the overlay file being merged in the Kconfig output: 

```
Parsing C:/git/golioth/modules/lib/golioth-firmware-sdk/examples/zephyr/hello/Kconfig
Loaded configuration 'C:/git/golioth/zephyr/boards/arm/nrf9160dk_nrf9160/nrf9160dk_nrf9160_ns_defconfig'
Merged configuration 'C:/git/golioth/modules/lib/golioth-firmware-sdk/examples/zephyr/hello/prj.conf'
Merged configuration 'C:/git/golioth/modules/lib/golioth-firmware-sdk/examples/zephyr/hello/boards/nrf9160dk_nrf9160_ns.conf'
```

rijesha, 2024-06-04:
@beriberikix did @rpiper suggestion solve your issue?

rijesha, 2024-06-21:
Closing this as stale. Please let me know if the issue persists.

closed 2024-06-21 by rijesha