mylonics/zephyr-ide #35: Initializing workspace before folder is selected

beriberikix, 2024-05-27:
Currently you can initialize a workspace when VS Code first launches and you haven't selected a folder. I've found myself keep on doing that by accident :) I've seen other extensions prevent you from taking any action before you select a folder (and providing a gentle reminder as a popup.)

Is this desired behavior? I can't think of a use case where it makes sense to initialize before a folder is selected.

rijesha, 2024-05-27:
It is undesired behaviour. In the future I was thinking there could be an option for a global zephyr install.

rijesha, 2024-05-27:
There is a prerelease version of the app that will be available v1.0.28. This version prevents initialization of the workspace before a folder in selected.

Please note that the new version no longer uses the debugOptimization variable and that by using the new version you will not be able to downgrade. (So save a copy of your projects in settings.json (or zephyr-ide.json)

closed 2024-05-29 by rijesha