mylonics/zephyr-ide #36: Add Project action buttons to primary bar

beriberikix, 2024-05-27:
It would be convenient to have action buttons for the active project displayed on the primary bar at the bottom. Once I've set up my project and build, I'm usually focused on the file explorer. I've found myself needing to switch back and forth between the extension UI to run a build (or use the command palette.)

Here's an example from PlatformIO
[image]

rijesha, 2024-05-27:
What all actions do you think should be available there?

rijesha, 2024-05-27:
There is a prerelease version of the app that will be available v1.0.28. That includes the GUI changes to add some buttons in the status menu.

Please note that the new version no longer uses the debugOptimization variable and that by using the new version you will not be able to downgrade. (So save a copy of your projects in settings.json (or zephyr-ide.json)

rijesha, 2024-06-04:
We can reopen this issue if there are any missing buttons.

closed 2024-06-04 by rijesha