mylonics/zephyr-ide #38: Too many errors detected by clang extension

NguyenTrongPhuc552003, 2024-05-31:
Hello, I am a user experiencing your extension. I followed your instructions and everything was fine from build to flash but for some reason I got errors like below:

[image]

- Zephyr SDK: ARM 0.16.6
- OS: Windows 11
- Board: NUCLEO-F411RE

If I delete that clang extension, there will be no more errors, but it also no longer gives me suggestions

Looking forward to hearing from you and see if it helps me. Thank !

NguyenTrongPhuc552003, 2024-05-31:
And now the Build is full of errors. In the terminal it says something dtsi file is missing that I don't know

[image]

rijesha, 2024-05-31:
This is not related to the extension or directly related to sw of the zephyr project. I would bring up the issue as a discussion point (not an issue) in the zephyr project and see how other users deal with it when using vscode. 
I personally don't use clang and just the CPP intellisense from microsoft.

If you find something interesting and that could be part of the extension setup please post again to this issue.

closed 2024-05-31 by rijesha