mylonics/zephyr-ide #44: Project management issues delete, add

zafersn, 2024-06-23:
Hi,

I have two different projects, I have a problem with the second project crpto, I can delete it from the project list, and I cant add build config. while having no problem with the board-test project,

Both of them are in the same workspace.

[image]

[image]

this error output: 

[image]

rijesha, 2024-06-23:
Also did you make the project with the current version or a previous version of the extension?

zafersn, 2024-06-23:
Hi @rijesha 

> Are you able to build the projects just fine? Are you able to share a repo of your workspace?
board-hw-test project is built fine. No problem. But when  I add another project, it wont be built, even though its in the same workspace the board-hw-test prj

> Those error outputs are not directly relevant to removing a project, but they shouldn't happen.
Well, actually, removing the project issue was because of me, I blocked the zephyr notf. so when I tried to delete it, it asked for permission through the notif. So thats why I missed them, and though I cant delete them,. 

> Also did you make the project with the current version or a previous version of the extension?

I am using v1.0.35 of it. is it uptodate?

rijesha, 2024-06-23:
Yea v1.0.35 is up to date. Could you reinitialize the workspace and see if it fixes it. Is the project visible in your settings.json file?

rijesha, 2024-06-24:
I just pushed a new version of v1.0.36.  Install the new version and run `Zephyr IDE: Debug Internal Shell` and post the contents here. You can do this by opening the command pallet and running the command. Then in the output window, select zephyr-ide and you will see the contents.

Ctrl-p -> Zephyr IDE: Debug Internal Shell

rijesha, 2024-08-01:
Closing this as stale. Let me know if the issue persists

closed 2024-08-01 by rijesha