mylonics/zephyr-ide #45: "external" folder is not updated after West Update (?)

joshegrimes, 2024-06-27:
Hi,

First off, just wanted to say I've been enjoying and appreciating using Zephyr IDE for STM32 development. I am still relatively inexperienced in Zephyr (and a college student), but it has been helpful in simplifying the learning process.

Recently, I've been working with a custom board that has a LED strip header (using a WS2812 strip). I kept getting build warnings related to that header even though I am 99% certain my device tree code, KConfig, etc. is solid. I looked into the devicetree bindings (specifically in external/zephyr/dts/bindings/led_strip) and found that my workspace is not the same (seemingly outdated) compared to the most recent version on the Zephyr Github page. For example, mine contains these .yaml files:

[image]

whereas theirs seems to be slightly revamped, especially with the strip I'm working with:

[image]

I tried running a West Update via the Zephyr IDE GUI, and that ran fine (said Zephyr 3.6.0 and everything...), but that did not seem to fix the issue. I am running the latest version (v1.0.36) of Zephyr IDE. I also tried going through the whole "Check Build Dependencies", "Setup West Environment", ..., "West Update" process once again and this also didn't seem to update those files. It may be worth noting that I'm using the "Local Zephyr Install" option. My workspace folder looks like this:

[image]

where "accel", "blinky", "bme280", "button_led", "button_led_multi", and "rospi_env_test" are all application folders.

So, not entirely sure if this is an actual issue with the VS Code extension or more of a gap in my understanding of west workspaces, but do you have any suggestions on how to resolve this? I also do recognize that updating these files may not fix my original build issues, but I thought starting here might be best.

Thanks in advance!

rijesha, 2024-06-27:
Im glad you are liking the app.

The latest version of zephyr (main/future 3.7.0) made quite a few changes with how the board files were organized. That may be why you are seeing something different on the local repo vs zephyr main.

Could you share some of the build warnings you are seeing?

joshegrimes, 2024-06-28:
Hi, appreciate the response.

I see.... that makes sense. Bit of a facepalm moment for me there. The build warnings I get consistently are:

```
warning: The choice symbol WS2812_STRIP_GPIO (defined at drivers/led_strip/Kconfig.ws2812:36) was
selected (set =y), but no symbol ended up as the choice selection. See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_WS2812_STRIP_GPIO and/or look up
WS2812_STRIP_GPIO in the menuconfig/guiconfig interface. The Application Development Primer, Setting      
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful       
too.
```
and
```
CMake Warning at C:/Users/joshe/Documents/Summer2024EmbeddedWork/STM32F4DISCO/external/zephyr/CMakeLists.txt:864 (message):
  No SOURCES given to Zephyr library: drivers__led_strip
```

I'm also getting build errors as well, but these warnings are what I'm most lost at resolving.

the defconfig file for my board is as such:
```
CONFIG_SOC_SERIES_STM32G4X=y
CONFIG_SOC_STM32G431XX=y

# enable uart driver
CONFIG_SERIAL=y

# enable GPIO
CONFIG_GPIO=y
CONFIG_GPIO_STM32=y

# enable clocks
CONFIG_CLOCK_CONTROL=y

# Console
CONFIG_CONSOLE=y
CONFIG_UART_CONSOLE=y

# Enable MPU
CONFIG_ARM_MPU=y

# Enable HW stack protection
CONFIG_HW_STACK_PROTECTION=y

# enable pin controller
CONFIG_PINCTRL=y

# led strip bus
CONFIG_LED_STRIP=y
CONFIG_WS2812_STRIP=y
CONFIG_WS2812_STRIP_GPIO=y

# can
CONFIG_CAN=y
CONFIG_CAN_FD_MODE=y
CONFIG_CAN_STM32_FDCAN=y

# adc
CONFIG_ADC=y
CONFIG_ADC_STM32=y
```
and my .dts file looks like this:
```

/dts-v1/;
#include <st/g4/stm32g431Xb.dtsi>
#include <st/g4/stm32g431c(6-8-b)ux-pinctrl.dtsi>
#include <zephyr/dt-bindings/input/input-event-codes.h>
#include <zephyr/dt-bindings/led/worldsemi_ws2812c.h>
#include <zephyr/dt-bindings/led/led.h>

/ {
	#address-cells = <1>;
	#size-cells = <1>;
	model = "ROSPi Env";
	compatible = "snsr,rospi_env";

	zephyr,user {
        io-channels = <&adc1 4>;
    };

	chosen {
		zephyr,console = &usart3;
		zephyr,shell-uart = &usart3;
		zephyr,sram = &sram0;
		zephyr,flash = &flash0;
		zephyr,canbus = &fdcan1;
	};

	aliases {
		led-strip = &led_strip;
	};

	led_strip: ws2812 {
		status = "okay";
        compatible = "worldsemi,ws2812-gpio";
        in-gpios = <&gpioa 4 GPIO_ACTIVE_HIGH>;
        chain-length = <8>;
		color-mapping = <LED_COLOR_ID_GREEN
				 		 LED_COLOR_ID_RED
				 		 LED_COLOR_ID_BLUE>;
    };
};

&clk_lsi {
	status = "okay";
};

/*
&clk_lse {
	status = "okay";
};
*/

&clk_hsi48 {
	status = "okay";
};

&clk_hse {
	clock-frequency = <DT_FREQ_M(16)>;
	status = "okay";
};

&pll {
	div-m = <4>;
	mul-n = <85>;
	div-p = <2>;
	div-q = <2>;
	div-r = <2>;
	clocks = <&clk_hse>;
	status = "okay";
};

&rcc {
	clocks = <&pll>;
	clock-frequency = <DT_FREQ_M(170)>;
	ahb-prescaler = <1>;
	apb1-prescaler = <1>;
	apb2-prescaler = <1>;
};

&gpioa {
	status = "okay";
};

&gpiob {
	status = "okay";
};

&gpioc {
	status = "okay";
};

/* gpiod, gpioe left out b/c ROSPi Env doesn't use them */

&gpiof {
	status = "okay";
};

&gpiog {
	status = "okay";
};

&adc1 {
	status = "okay";
    #address-cells = <1>;
    #size-cells = <0>;

	pinctrl-0 = <&adc1_in4_pa3>;
    pinctrl-names = "default";
	st,adc-clock-source = <1>;
	st,adc-prescaler = <1>;

    channel@4 {
		reg = <4>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
		zephyr,resolution = <12>;
	};
};

&i2c2 {
	pinctrl-0 = <&i2c2_scl_pa9 &i2c2_sda_pa8>;
	pinctrl-names = "default";
	status = "okay";

	light_sens: opt3001@44 {
        compatible = "ti,opt3001";
        reg = <0x44>;
    };

	barometer: lps22hb@5c {
		compatible = "st,lps22hb-press";
		reg = <0x5c>;
	};

	env_sens: bme680@76 {
		compatible = "bosch,bme680";
		reg = <0x76>;
	};

};

&fdcan1 {
	pinctrl-0 = <&fdcan1_tx_pa12 &fdcan1_rx_pa11>;
	pinctrl-names = "default";
	status = "okay";

	bus-speed = <500000>;
	bus-speed-data = <500000>;
};

/* uart4 is what's listed on the schematic, but usart3 is what corresponds
   to the pins...
 */ 
&usart3 { 
    status = "okay";
	pinctrl-0 = <&usart3_tx_pc10 &usart3_rx_pc11>;
	pinctrl-names = "default";
	current-speed = <115200>;
};
```
I am not using a .overlay file or anything. We are currently trying to make a default setup for this custom board to be used with any application that's why I'm only working in the defconfig and board's .dts.

Feel free to look into this, or not, but please don't feel _obligated_ to help with these warnings. Clearly this is not an issue with the extension and not the correct thread to be posting on for help.

joshegrimes, 2024-06-28:
Okay, so I realized that after looking closer at the dependencies for each KConfig line, it looks like doing the WS2812 data over GPIO is currently only allowed for NRF51 SOCs and not STM32 with the Zephyr drivers. So don't worry about looking into this, we will find another option to write our data.

[image]

Thanks again!

rijesha, 2024-06-28:
Glad you figured it out

closed 2024-06-28 by rijesha