mylonics/zephyr-ide #48: Extension doesn't show in sidebar

ronsegal, 2024-07-22:
On Ubuntu 24.02 with all required zephyr getting started extensions installed and version verified including python3 venv, Zephyr IDE vscode extension appears to install without error but does not appear in the task sidebar. Other IDE extensions do appear, platformio etc. Have tried uninstalling re-installing restarting vscode etc but can't the Zephyr IDE extension to appear.  Suggestions appreciated.

rijesha, 2024-07-22:
Could you try an earlier version of the extension like 1.0.35 and see if that works.
If you right click on the activity bar is zephyr ide an option there?

ronsegal, 2024-07-23:
Thanks for getting back.  I've tried both of two versions before the latest, still with no Zephyr IDE available in the left hand activity bar at all, hidden or otherwise.  The task bar at the bottom does contain Zephyr IDE functional icons, also the command pallet shows Zephyr IDE commands.

djr-spectrummfg, 2024-07-23:
I had this problem, I was able to fix it by deleting the entire extensions folder (`%USERPROFILE%\.vscode\extensions` or `~/.vscode/extensions`) and installing just the zephyr IDE plugin. Probably this is overkill though.

rijesha, 2024-07-23:
I'm not sure what is causing this problem, but I will have a look into it further when I have some time.

ronsegal, 2024-07-23:
Thanks djr and Rijesh.  On your suggestion djr I removed all extensions and started again (perhaps a good idea anyway) with a blank sheet.  Zephyr IDE now shows in the sidebar.  Suggest that's more of a workaround than solving the underlying problem, some kind of clash with another extension.  Will get back if I manage to discover which one(s), although that might of course depend on installation order.

ronsegal, 2024-07-24:
Further. By the look the clash is with the Python Environment Manager extension plus the Python extension being mutually exclusive with Zephyr IDE on the activity bar.  Sorry no time to investigate further, though obviously a significant issue with those Python extensions being key ones.

gBonvenuto, 2024-08-15:
I believe this should be noted in the Know Issues on the README until we have a fix

rijesha, 2024-08-23:
@gBonvenuto feel free to make a PR.

rijesha, 2024-08-23:
@ronsegal Why do you think it is related to python extensions or do you think it is just that your local python extension was causing a conflict?

rijesha, 2024-08-23:
If anyone runs into the issue again could you post any error messages from the developer console?
F1 > Developer: Toggle Developer Tools > Console

rijesha, 2024-08-23:
I found the issue. It was specifically with the python environment manager extension. Thanks @ronsegal your findings really helped. 
This is all that was needed to change. 0b3f8af329d6a33d6d462530e995e302472836bf

closed 2024-08-23 by rijesha

rijesha, 2024-08-23:
Also bug report added to the python environment manager extension. https://github.com/DonJayamanne/vscode-python-manager/issues/94

ronsegal, 2024-08-23:
Well done, very much appreciate the fix.