mylonics/zephyr-ide #51: Configure C++ intellisense automatically

beriberikix, 2024-08-09:
https://github.com/beriberikix/zephyr-vscode-example/blob/main/.vscode/settings.json#L6-L7

The official [C++ Extension](https://marketplace.visualstudio.com/items?itemName=ms-vscode.cpptools) from Microsoft can be configured for intellisense by setting the compiler path and compile commands. The extension likely knows this information and could easily set it automatically.

rijesha, 2024-08-11:
Any thoughts on how to deal with multiple builds/ multiple compile commands.

As a reference here are a couple of answers on the web.
https://stackoverflow.com/questions/77517796/intellisense-configuration-for-multiple-builds
https://stackoverflow.com/questions/77394438/merge-multiple-compile-command-json-files

The first reference mentions using cmake, which I am not sure is viable. 
The second reference just merges all the json files into one. Which I am also not sure if that will just work.

rijesha, 2024-08-11:
This is how the ESP app does it and may be the way to go.
https://github.com/espressif/vscode-esp-idf-extension/blob/81dea08d84874231a6250f946d2cc0d21b028ceb/templates/.vscode/c_cpp_properties.json#L6

beriberikix, 2024-08-12:
Ah, good question! I've only used single build at this point. Interesting problem :thinking:

rijesha, 2024-08-24:
Implemented in v1.0.43.

closed 2024-08-24 by rijesha