mylonics/zephyr-ide #54: west init fails

MathCunha10, 2024-08-21:
```
 *  Executing task: west init -l C:\Users\Matheus\.zephyr_ide\application  

FATAL ERROR: can't init: no west.yml found in C:\Users\Matheus\Baja\UsersMatheus.zephyr_ideapplication
```
Fails on both global and local zephyr install 

v1.0.40

rijesha, 2024-08-22:
Is this a fresh install of zephyr-ide or did you upgrade from previous versions? 
It seems that the path separator is not being added for you. 
What version of vscode and windows are you using?

rijesha, 2024-08-26:
A related issue was found and fixed in 1.1.1. Hopefully that resolves this problem as well. We can reopen the issue if you are still having this problem.

closed 2024-08-26 by rijesha

rijesha, 2024-08-28:
I think I may have found a related cause. If you have bash set as your default terminal in windows this can happen. You need to use powershell as the default.