mylonics/zephyr-ide #59: Multiple KConfig files not found

rokobertcm, 2024-09-03:
If I include multiple extra .conf file in project kConfig error "File not found" is generated.

"/bin/bash '-l', '-c', 'west build -b ECO00189 */Soft/zpIDE3/B300 -p --build-dir /Users/robert.konc/Soft/zpIDE3/B300/build/ECO00189 -DBOARD_ROOT='*/Soft/zpIDE3' -DEXTRA_CONF_FILE='*/Soft/zpIDE3/B300/prj_debug.conf,*/Soft/zpIDE3/B300/prj_ble.conf' '" terminated with exit code: 1.

EXTRA_CONF_FILE macro could include multiple files but must be separated with (Space) or (;). Here is used (,).

rijesha, 2024-09-04:
implemented in 1.3.0. Please let me know if the issue is fixed

rokobertcm, 2024-09-04:
With version 1.3 this bug is resolved.
Great job. Thank you!

closed 2024-09-04 by rokobertcm