mylonics/zephyr-ide #62: Add command:zephyr-ide.get-active-build-board-path

andre-stefanov, 2024-09-28:
In order to configure cortex-debug + openocd launch configuration in launch.json one has to provide the openocd.cfg file (otherwise cortex-debug won't start openocd):

```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Zephyr IDE: Debug",
            "cwd": "${workspaceFolder}",
            "gdbPath": "${command:zephyr-ide.get-gdb-path}",
            "executable": "${command:zephyr-ide.get-active-build-path}/zephyr/zephyr.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "showDevDebugOutput": "raw",
            "interface": "swd",
            "configFiles": [
                "${workspaceFolder}/external/zephyr/boards/st/nucleo_f446re/support/openocd.cfg"
            ],
        }
    ]
}
```

This is a working configuration for a nucleo_f446re board. But as soon as one wants to debug on another board, one has to provide another path. Since the location of the openocd.cfg file is pretty much the same relatively to the board directory ({boardDir}/support.cfg) one could change the value to:

```
"${command:zephyr-ide.get-active-build-board-path}/support/openocd.cfg"
```

This way the same command could be reused for many builds and boards (as long as they support openocd).

rijesha, 2024-09-29:
Should be fairly straight forward to implement. Thanks for the issue.

rijesha, 2024-09-29:
Could you test out the latest pre-release (1.3.3) and see if this works for you? 

`"${command:zephyr-ide.get-active-build-board-path}/support/openocd.cfg"`

andre-stefanov, 2024-09-29:
wow thanks for the quick implementation! i've just tested it with a local built extension and it works like a charm:

```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Zephyr IDE: Debug",
            "cwd": "${workspaceFolder}",
            "gdbPath": "${command:zephyr-ide.get-gdb-path}",
            "executable": "${command:zephyr-ide.get-active-build-path}/zephyr/zephyr.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "interface": "swd",
            "configFiles": [
                "${command:zephyr-ide.get-active-build-board-path}/support/openocd.cfg"
            ]
        },
    ]
}
```
results in
```bash
openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/andre/repos/zephyr-ide-test-update -f /home/andre/.vscode/extensions/marus25.cortex-debug-1.12.1/support/openocd-helpers.tcl -f /home/andre/repos/zephyr-ide-test-update/external/zephyr/boards/st/nucleo_f446re/support/openocd.cfg
```

rijesha, 2024-09-29:
Great. Thanks for testing. 
FYI in the vscode marketplace, there is a prerelease stream you can enable per extension. On that stream 1.3.3 is available. So in the future you don't necessarily need to locally build the extension.

closed 2024-09-29 by rijesha