mylonics/zephyr-ide #63: Provide an example project repository

andre-stefanov, 2024-09-28:
At the moment i am struggling to setup a zephyr-ide project repository because of following issues/questions:

- which files have to be checked into the repository and which should be ignored
- some properties/settings get overriden to absolute paths after a reinitialization 
  - devicetree.zephyr  #61 
  - kconfig.zephyr.base #61 
  - relBoardSubDir (in zephyr-ide.json) is being set to absolute path. E.g. `"relBoardSubDir": "/home/andre/repos/my-project/external/zephyr/boards/st/nucleo_f446re"`
 - how to properly clone and setup an existing repository
   - first checkout, open in vscode and then initialize?
   - initialize from repository? This fails for me with `FATAL ERROR: can't init: no west.yml found in /home/andre/repos/my-project/.west/manifest-tmp`
 
An example application like https://github.com/zephyrproject-rtos/example-application would be very helpful to make usage of this extension much easier.

rijesha, 2024-09-29:
I can look into this, I am thinking of automatically generating a .gitignore file. 
Currently you should only commit launch.json, vscode.json, .west/config, and project directories.

Can you provide the repository where this failed?
"initialize from repository? This fails for me with FATAL ERROR: can't init: no west.yml found in /home/andre/repos/my-project/.west/manifest-tmp"

andre-stefanov, 2024-09-29:
nice! i've included files you mentioned and it worked. I also added `application/west.yml` because it defines required zepyhr version (v3.7.0 in my case).

andre-stefanov, 2024-09-29:
As of the issue i mentioned:

1. open vscode in an empty folder
2. in "extension setup" click on "workspace"
3. initialize workspace
4. from git repo
5. use https://github.com/andre-stefanov/zephyr-ide-test.git

```bash
=== Initializing in /tmp/zephyr-ide-test
--- Cloning manifest repository from https://github.com/andre-stefanov/zephyr-ide-test.git
Cloning into '/tmp/zephyr-ide-test/.west/manifest-tmp'...
remote: Enumerating objects: 28, done.
remote: Counting objects: 100% (28/28), done.
remote: Compressing objects: 100% (17/17), done.
remote: Total 28 (delta 5), reused 28 (delta 5), pack-reused 0 (from 0)
Receiving objects: 100% (28/28), 4.34 KiB | 4.34 MiB/s, done.
Resolving deltas: 100% (5/5), done.
FATAL ERROR: can't init: no west.yml found in /tmp/zephyr-ide-test/.west/manifest-tmp
  Hint: check --manifest-url=https://github.com/andre-stefanov/zephyr-ide-test.git  You may need to remove /tmp/zephyr-ide-test/.west before retrying.

 *  The terminal process "/usr/bin/zsh '-c', 'west init -m https://github.com/andre-stefanov/zephyr-ide-test.git '" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
 ```

rijesha, 2024-10-30:
Added automatic .gitignore generation.
Added a sample project and referenced it in manual and readme.

Thanks for the issue

closed 2024-10-30 by rijesha