mylonics/zephyr-ide #67: Add latest v3.7.0 to install menu when initializing workspace

anthonyhoegberg, 2024-10-09:
Maybe add the latest 3.7.0 when initializing workspace
https://github.com/zephyrproject-rtos/zephyr/releases

Also, i would like if there could be support for custom SDK version selection? 
I need to use SDK release candidate but i cant use it from the ide directly when setting up SDK etc.

rijesha, 2024-10-09:
Adding 3.7.0 is definitely doable.
Importing a custom SDK is a bit more difficult. An MD5 file is used to download the sdk, does the release candidate have an MD5 file ready?

anthonyhoegberg, 2024-10-10:
> Adding 3.7.0 is definitely doable.

Thanks
> Adding 3.7.0 is definitely doable.
Importing a custom SDK is a bit more difficult. An MD5 file is used to download the sdk, does the release candidate have an MD5 file ready?

That i dont know

rijesha, 2024-10-10:
ok. You can try just manually installing the toolchain into the toolchain directory. It is usually located in your home/main directory in a folder called .zephyr_ide

rijesha, 2024-10-30:
Adding v3.7.0 is commited and will be available in next release. Thanks for the issue.

closed 2024-10-30 by rijesha