mylonics/zephyr-ide #68: Open terminal with enviroment?

anthonyhoegberg, 2024-10-09:
It would be nice it you guys could add an option to open the terminal with the virtual env already loaded so that we manually dont have to load the .venv and can just use our tools or do commands normally from there

rijesha, 2024-10-09:
I think this feature is already available. You can open a Zephyr IDE terminal.

When you open a zephyr IDE terminal, can you run the following commands and see where the python executable is pointing at.

```
python
import sys
sys.executable
```

anthonyhoegberg, 2024-10-10:
Ahh i found it under the terminal options menu, i was looking around in the command palette and i couldn't find it in there

DBS06, 2024-10-10:
@rijesha Maybe it is a dump question, but is it possible to run custom tasks inside a "Zephyr IDE terminal"?
Because somehow, I am not able to run i.e. ```west build [...]``` from a task or start the zephyr virtual env.
I need to do this to run renode from vscode and I want to run a build task prior to start renode. Furthermore I created a task to get a ROM and RAM Report, right now I am using a workaround to avoid using the west command, but it would be nice to run it.
The RAM/ROM report could be a very useful feature for Zephyr-IDE. Sadly I am not familiar with Typescript, because otherwise I would add this feature an create a PR.

closed 2024-10-30 by rijesha

rijesha, 2024-10-30:
@DBS06 Here is the commit to show how the ram/rom report feature was added. #39adf0

DBS06, 2024-10-30:
Perfect, looks awesome 👍