mylonics/zephyr-ide #70: Add command to get the current used board

DBS06, 2024-10-22:
First of all you have done a great Job with this Plugin!!!

Similar to these one:

    "zephyr-ide.get-active-project-name"
    "zephyr-ide.get-active-project-path"
    "zephyr-ide.get-active-build-path"
    "zephyr-ide.get-active-build-board-path"
    "zephyr-ide.select-active-build-path"
    "zephyr-ide.get-gdb-path"
    "zephyr-ide.get-toolchain-path"

something like:

    "zephyr-ide.get-active-board-name" -> name of the board in zephyr
    "zephyr-ide.get-active-device-name" -> name of the device for debugger (needed for JLink) maybe this needs an extra field in the 'zephyr-ide.json'

would be very nice.
If you could give me a hint I could try to implement it by myself and provide a PR, but I have no experience at all in Typescript, therefore I am asking first.

rijesha, 2024-10-24:
Hi thanks for the issue. 
Could you give an example of what the active device name looks like? Should this be a custom field for each build?

DBS06, 2024-10-29:
First, sorry for my late reply.
Examples of a active device name for SEGGER-JLink are:
`NRF52840_XXAA, STM32F401RE`

Why do I need this or I think it is useful:
To have a generic Debug-Configuration for different MCUs in `launch.json` with JLink, I need to specify which MCU is used and where the corresponding svd-File is. There are two options:
- Create a Debug-Configuration for each target
  - advantage: I don't have to select for each build configuration the target
  - disadvantage: for each build configuration I need to set the "Debug", "Build Debug" and "Debug Attach" commands for Zephyr-IDE for every new build-configuration and target. Furthermore, the entries in `launch.json` are growing with each target.
- Create an `"inputs"` entry of type `"type": "pickString",` in `launch.json` where I can select the device and svd-file for JLInk
  - advantage: the `launch.json` only has the 3 entries for `Zephyr IDE: Debug Select`, `Zephyr IDE: Debug` and `Zephyr IDE: Attach` and I don't have to manually specify for each build-configuration the  "Debug", "Build Debug" and "Debug Attach" commands.
  - disadvantage: For every debug sessions I have to select the JLink-Target and SVD-File.
  - Here you can see how I am currently doing it: [launch.json](https://github.com/user-attachments/files/17553807/launch.json)
 
With an option to get `zephyr-ide.get-active-board-name` and `zephyr-ide.get-active-device-name` I have the advantages of both options and not the disadvantages.

rijesha, 2024-10-29:
That makes sense. I have been wondering how to include svd files. Where is your svd file usually located? Do you know if the zephyr project includes them? 
What would be nice is a list of boards, with their corresponding SVD file, but I don't know if that exists or how zephyr users deal with SVD files in general.

DBS06, 2024-10-30:
In general the SVD Files which I need for my targets are in the project folder.
I usually search for them on the internet and download them.

rijesha, 2024-11-02:
I have added a feature for defining custom variables in your zephyr-ide.json file. It will be up in prerelease 1.3.5 which will be up later today.

If there is a variable you want associated with a project/build that needs to be available for your launch configuration, you can use the `zephyr-ide.get-active-project-variable` or `zephyr-ide.get-active-build-variable`. In you `zephyr-ide.json` file, create a `vars` variable in your project or the buildConfig and define a custom variable. Then in launch.json you can access the variable using the input command. 

zephyr-ide.json
```
{
  "projects": {
    "blinky": {
      "name": "blinky",
      "vars": {
        "custom_var": "custom_var"
      },
      "buildConfigs": {
        "build\\stm32f4_disco": {
          "relBoardDir": "external\\zephyr\\boards",
          "board": "stm32f4_disco",
          "relBoardSubDir": "external\\zephyr\\boards\\st\\stm32f4_disco",
          "vars": {
            "jlink_var": "STM32F401RE",
            "bmp_port": "COM3"
          },
...
```

launch.json
```
"inputs": [
        {
            "id": "getCustomBuildVariable",
            "type": "command",
            "command": "zephyr-ide.get-active-build-variable",
            "args": "bmp_port"
        }
    ],
"configurations": [    
      {
        "name": "Zephyr IDE: Debug",
        "BMPGDBSerialPort": "${input:getCustomBuildVariable}",
      },
    ],
```

DBS06, 2024-11-04:
That's genius! Thanks a lot!

closed 2024-11-04 by rijesha

DBS06, 2024-11-04:
Thanks a lot for the compliment about my photography 😉
I also rated the extension.
IMHO it is underrated.
I also recommended it on the zephyr discord channel a couple of days ago.

StephenPPEng, 2025-03-25:
@rijesha  could you add documentation/examples somewhere in the manual for this behavior?  I spent about 4 hours trying to implement something similar in VS code because i couldn't find examples for this support anywhere.
Additionally, adding better documentation/etc for all of the ${command:zephyr-ide.**} commands?
Love the extension and your support for it.

rijesha, 2025-03-26:
@StephenPPEng If you have any direct suggestions to improvements onto to the manual feel free to make a PR to update the manual or adding an example. 
https://github.com/mylonics/zephyr-ide/blob/main/docs/MANUAL.md

The custom variable command is documented in the manual, but if there is confusion with the way it is documented let me know.