mylonics/zephyr-ide #71: Run custom tasks from zephyr ide terminal

rijesha, 2024-10-30:
@DBS06 had the following request:

Maybe it is a dump question, but is it possible to run custom tasks inside a "Zephyr IDE terminal"?
Because somehow, I am not able to run i.e. west build [...] from a task or start the zephyr virtual env.
I need to do this to run renode from vscode and I want to run a build task prior to start renode. Furthermore I created a task to get a ROM and RAM Report, right now I am using a workaround to avoid using the west command, but it would be nice to run it.
The RAM/ROM report could be a very useful feature for Zephyr-IDE. Sadly I am not familiar with Typescript, because otherwise I would add this feature an create a PR.

Btw: You guys have done a really good job with this extension and I am really thankful!!!

Edit: I used a freestanding application and with this configuration it does not work well, but nevertheless I am using a workspace configuration now

DBS06, 2024-10-30:
I had this problem with a freestanding application, with Zephyr integrated in the project folder (external/zephyr), I had no problem with it.

Besides the RAM/ROM Report, I also created a task for dtsh to get the device tree nicely visualized on the terminal. Somehow I am not able to use the Nordic -Devicetree vscode extension without the nRF-Connect-SDK.

rijesha, 2024-10-30:
What is the command you would use for dtsh?

rijesha, 2024-10-30:
I added a dtsh command as well.

rijesha, 2024-10-30:
I think the tasks should already run in the default terminal. See here https://code.visualstudio.com/docs/terminal/profiles#_configuring-the-taskdebug-profile.
What is in your settings.json file for the default terminal profile.
Mine says: "terminal.integrated.defaultProfile.windows": "Zephyr IDE Terminal"

DBS06, 2024-11-04:
Thanks a lot!

closed 2024-11-04 by rijesha