mylonics/zephyr-ide #75: Bug in getShellEnvironment function

rokobertcm, 2024-11-07:
Every time getShellEnvironment function is called variable envPath["PATH"] is bigger.

Current  code:
```
if (setupState.env["PATH"]) {
  envPath["PATH"] = path.join(setupState.env["PATH"], pathdivider + envPath["PATH"]);
}
```

My solution:
```
if (setupState.env["PATH"]) {
  let envstr = setupState.env["PATH"].replace("/:", ""); // do not need / path
  if (!envPath["PATH"]?.includes(envstr)) {
    envPath["PATH"] = envstr + envPath["PATH"]; // path.join(setupState.env["PATH"], pathdivider + envPath["PATH"].);
  }
}
```

It was tested on MacOS platform.

@rijesha instead of test if envstr is in envPath["PATH"] you could implement new variable which will indicate that setupState.env["PATH"] is changed and only then update envPath["PATH"].
If you agree you could implement in next release.

rijesha, 2024-11-12:
Hi Robert. Thanks for the issue. Your solution is probably that best one. It is hard to implement a variable to indicate when it has been updated. Some form of preventing this issue will be included in the next release.

rokobertcm, 2024-11-19:
Super. Thank you!

rijesha, 2024-11-22:
Implemented in prerelease 1.4.1

closed 2024-11-22 by rijesha