mylonics/zephyr-ide #76: Select specific device before flash

aqzi, 2024-11-09:
When I connect multiple devices to my computer, the zephyr IDE always flashes to the most recently connected device. Is there a way to select a specific device before flashing?

rijesha, 2024-11-12:
Can you give an example of what devices you plug in and what runner you are using for flashing.

aqzi, 2024-11-12:
I’m working with three STM32 Nucleo-F429ZI boards and OpenOCD as runner. I currently limit the setup to a single connected device at a time. However, this setup is inconvenient as it requires frequent switching of the boards.

rijesha, 2024-11-14:
I think this is more of an OpenOCD type of configuration. I would setup up multiple OpenOCD runners, with each runner having runner arguments that point to the specific hardware.
Here is a a stack overflow question on setting something like that up. There is an hla_serial that may allow you to specify the hardware.

aqzi, 2024-11-14:
Ok thx I will try that and what is the link of the stack overflow question?

rijesha, 2024-11-14:
https://stackoverflow.com/questions/48211067/how-to-use-openocd-to-talk-to-two-stm32-boards-at-once

closed 2024-11-14 by aqzi