mylonics/zephyr-ide #84: Build command

rokobertcm, 2024-11-26:
Is it possible to adapt build command that includes only source build folder and --build-dir folder.
I think other options like b- (board), and any -D(options) are only relevant in pristine build (-p).
Compile time is faster. And I think it is not necessary each time reconfigure cmake settings and so on.

This is example.

Pristine build:
`west build -b frdm_mcxw71 /Users/robert.konc/Soft/zpIDE3/blinky -p --build-dir /Users/robert.konc/Soft/zpIDE3/blinky/build/frdm_mcxw71  -- -DBOARD_ROOT='/Users/robert.konc/Soft/zpIDE3/external/zephyr'`

Build:
`west build -b frdm_mcxw71 /Users/robert.konc/Soft/zpIDE3/blinky  --build-dir /Users/robert.konc/Soft/zpIDE3/blinky/build/frdm_mcxw71  -- -DBOARD_ROOT='/Users/robert.konc/Soft/zpIDE3/external/zephyr'`

Build, my proposal:
`west build /Users/robert.konc/Soft/zpIDE3/blinky  --build-dir /Users/robert.konc/Soft/zpIDE3/blinky/build/frdm_mcxw71`

rijesha, 2024-11-28:
This is great. I have pushed the change in v1.5.0

rokobertcm, 2024-11-29:
Thank you for implementing this option.
But is still problem if I have extra cons file in build arguments. Adding extra conf files make build as pristine even without -p. 
Please use only project directory and build directory in west build arguments and it will be ok.

rijesha, 2024-11-29:
Ok I pushed a new update.

rokobertcm, 2024-11-30:
Sorry for bother you again.
It still not work.
In attached file is paths which I was create with my solution.
[patch.diff.zip](https://github.com/user-attachments/files/17966825/patch.diff.zip)

rijesha, 2024-11-30:
Ok. I see that I missed a bit in my change. Could you try it again? If it doesn't work can you make a PR with your fix. It will be easier for me to review.

rijesha, 2024-12-12:
Could you check if the latest zephyr ide version works for you. If it doesn't let me know. Otherwise I think this issue can be closed.

rijesha, 2024-12-17:
Closing this issue as implemented. We can reopen if there are issues.

closed 2024-12-17 by rijesha