mylonics/zephyr-ide #85: Changes to Windows path are not reflected in Zephyr IDE Terminal

glenn-andrews, 2024-11-28:
I installed OpenOCD, which required adding the xPack directory to my path. 

From a PowerShell window:
```
$env:Path        
C:\Python312\Scripts\;C:\Python312\;C:\Program Files (x86)\STMicroelectronics\stlink_server;C:\ST\STM32CubeCLT_1.16.0;C:\ST\STM32CubeCLT_1.16.0\CMake\bin;C:\ST\STM32CubeCLT_1.16.0\Ninja\bin;C:\ST\STM32CubeCLT_1.16.0\GNU-tools-for-STM32\bin;C:\ST\STM32CubeCLT_1.16.0\STLink-gdb-server\bin;C:\ST\STM32CubeCLT_1.16.0\STM32CubeProgrammer\bin;C:\Program Files\ImageMagick-7.1.1-Q16-HDRI;C:\Program Files\Eclipse Adoptium\jdk-21.0.3.9-hotspot\bin;C:\Python311\Scripts\;C:\Python311\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Microsoft VS Code\bin;C:\ProgramData\chocolatey\bin;C:\Program Files\CMake\bin;C:\Program Files\Git\cmd;C:\Program Files\LLVM\bin;C:\Program Files\doxygen\bin;C:\Strawberry\c\bin;C:\Strawberry\perl\site\bin;C:\Strawberry\perl\bin;C:\Program Files\MiKTeX\miktex\bin\x64\;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin\;C:\Program Files\TortoiseGit\bin;C:\Program Files\nodejs\;C:\Users\glenn\AppData\Local\Microsoft\WindowsApps;C:\Program Files\OpenOCD\bin;C:\Users\glenn\AppData\Local\GitHubDesktop\bin;C:\Program Files\qemu;C:\Users\glenn;C:\msys64\usr\bin;C:\Users\glenn\AppData\Roaming\npm;C:\Users\glenn\AppData\Roaming\xPacks\@xpack-dev-tools\openocd\0.12.0-4.1\.content\bin\;
```

From the Zephyr IDE Terminal Window:
```
C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Program Files (x86)\STMicroelectronics\stlink_server;C:\ST\STM32CubeCLT_1.16.0;C:\ST\STM32CubeCLT_1.16.0\CMake\bin;C:\ST\STM32CubeCLT_1.16.0\Ninja\bin;C:\ST\STM32CubeCLT_1.16.0\GNU-tools-for-STM32\bin;C:\ST\STM32CubeCLT_1.16.0\STLink-gdb-server\bin;C:\ST\STM32CubeCLT_1.16.0\STM32CubeProgrammer\bin;C:\Program Files\ImageMagick-7.1.1-Q16-HDRI;C:\Program Files\Eclipse Adoptium\jdk-21.0.3.9-hotspot\bin;C:\Python311\Scripts\;C:\Python311\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Microsoft VS Code\bin;C:\ProgramData\chocolatey\bin;C:\Program Files\CMake\bin;C:\Program Files\Git\cmd;C:\Program Files\LLVM\bin;C:\Program Files\doxygen\bin;C:\Strawberry\c\bin;C:\Strawberry\perl\site\bin;C:\Strawberry\perl\bin;C:\Program Files\MiKTeX\miktex\bin\x64\;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin\;C:\Program Files\TortoiseGit\bin;C:\Users\glenn\AppData\Local\Microsoft\WindowsApps;C:\Program Files\OpenOCD\bin;C:\Users\glenn\AppData\Local\GitHubDesktop\bin;C:\Program Files\qemu;C:\Users\glenn;
```

I've tried resetting the workspace settings. Do I need to delete and reinstall the extension?

glenn-andrews, 2024-11-28:
Removing and reinstalling does refresh the path.

rijesha, 2024-11-28:
No you dont need to reinstall the extension. Just restarting VSCode should work. You can experiment and see when the path variable updates in vscode when you change it externally. Making a new terminal instance may update it but Im not sure.

glenn-andrews, 2024-11-29:
I tried restarting vscode and it remained the same.

rijesha, 2024-11-29:
Zephyr ide just appends to whatever path is available in the terminal.
Open vs code and start a PowerShell terminal. Then list the environment variables. Does the new path show up there?
If you open a zephyr ide terminal and do the same things does the new path show up there?

glenn-andrews, 2024-11-29:
When I was testing it, the powershell in vscode showed the correct path, and the Zephyr IDE shell did not.

Is it because one is running on the virtual environment?

rijesha, 2024-11-29:
No. It just has to do with when the variable is registered.

Let's do a quick test with your system.

Please add a variable to path and then open vscode. 
Now check if the variable shows up in the zephyr-ide shell and if it shows up in powershell and report back.

glenn-andrews, 2024-11-29:
Added `C:\oolite` to the path and opened VSCode:

PowerShell:
```
C:\Python312\Scripts\;C:\Python312\;C:\Program Files (x86)\STMicroelectronics\stlink_server;C:\ST\STM32CubeCLT_1.16.0;C:\ST\STM32CubeCLT_1.16.0\CMake\bin;C:\ST\STM32CubeCLT_1.16.0\Ninja\bin;C:\ST\STM32CubeCLT_1.16.0\GNU-tools-for-STM32\bin;C:\ST\STM32CubeCLT_1.16.0\STLink-gdb-server\bin;C:\ST\STM32CubeCLT_1.16.0\STM32CubeProgrammer\bin;C:\Program Files\ImageMagick-7.1.1-Q16-HDRI;C:\Program Files\Eclipse Adoptium\jdk-21.0.3.9-hotspot\bin;C:\Python311\Scripts\;C:\Python311\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Microsoft VS Code\bin;C:\ProgramData\chocolatey\bin;C:\Program Files\CMake\bin;C:\Program Files\Git\cmd;C:\Program Files\LLVM\bin;C:\Program Files\doxygen\bin;C:\Strawberry\c\bin;C:\Strawberry\perl\site\bin;C:\Strawberry\perl\bin;C:\Program Files\MiKTeX\miktex\bin\x64\;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin\;C:\Program Files\TortoiseGit\bin;C:\Program Files\nodejs\;C:\Users\glenn\AppData\Local\Microsoft\WindowsApps;C:\Program Files\OpenOCD\bin;C:\Users\glenn\AppData\Local\GitHubDesktop\bin;C:\Program Files\qemu;C:\Users\glenn;C:\msys64\usr\bin;C:\Users\glenn\AppData\Roaming\npm;C:\Users\glenn\AppData\Roaming\xPacks\@xpack-dev-tools\openocd\0.12.0-4.1\.content\bin\;C:\Oolite;
```

Zephyr IDE Shell:
```
C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Python312\Scripts\;C:\Python312\;C:\Program Files (x86)\STMicroelectronics\stlink_server;C:\ST\STM32CubeCLT_1.16.0;C:\ST\STM32CubeCLT_1.16.0\CMake\bin;C:\ST\STM32CubeCLT_1.16.0\Ninja\bin;C:\ST\STM32CubeCLT_1.16.0\GNU-tools-for-STM32\bin;C:\ST\STM32CubeCLT_1.16.0\STLink-gdb-server\bin;C:\ST\STM32CubeCLT_1.16.0\STM32CubeProgrammer\bin;C:\Program Files\ImageMagick-7.1.1-Q16-HDRI;C:\Program Files\Eclipse Adoptium\jdk-21.0.3.9-hotspot\bin;C:\Python311\Scripts\;C:\Python311\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Microsoft VS Code\bin;C:\ProgramData\chocolatey\bin;C:\Program Files\CMake\bin;C:\Program Files\Git\cmd;C:\Program Files\LLVM\bin;C:\Program Files\doxygen\bin;C:\Strawberry\c\bin;C:\Strawberry\perl\site\bin;C:\Strawberry\perl\bin;C:\Program Files\MiKTeX\miktex\bin\x64\;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin\;C:\Program Files\TortoiseGit\bin;C:\Program Files\nodejs\;C:\Users\glenn\AppData\Local\Microsoft\WindowsApps;C:\Program Files\OpenOCD\bin;C:\Users\glenn\AppData\Local\GitHubDesktop\bin;C:\Program Files\qemu;C:\Users\glenn;C:\msys64\usr\bin;C:\Users\glenn\AppData\Roaming\npm;C:\Users\glenn\AppData\Roaming\xPacks\@xpack-dev-tools\openocd\0.12.0-4.1\.content\bin\;
```

I think I found it. That was the shell open when I opened VSCode. Opening a second shell shows the correct output:
```
C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Python312\Scripts\;C:\Python312\;C:\Program Files (x86)\STMicroelectronics\stlink_server;C:\ST\STM32CubeCLT_1.16.0;C:\ST\STM32CubeCLT_1.16.0\CMake\bin;C:\ST\STM32CubeCLT_1.16.0\Ninja\bin;C:\ST\STM32CubeCLT_1.16.0\GNU-tools-for-STM32\bin;C:\ST\STM32CubeCLT_1.16.0\STLink-gdb-server\bin;C:\ST\STM32CubeCLT_1.16.0\STM32CubeProgrammer\bin;C:\Program Files\ImageMagick-7.1.1-Q16-HDRI;C:\Program Files\Eclipse Adoptium\jdk-21.0.3.9-hotspot\bin;C:\Python311\Scripts\;C:\Python311\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Microsoft VS Code\bin;C:\ProgramData\chocolatey\bin;C:\Program Files\CMake\bin;C:\Program Files\Git\cmd;C:\Program Files\LLVM\bin;C:\Program Files\doxygen\bin;C:\Strawberry\c\bin;C:\Strawberry\perl\site\bin;C:\Strawberry\perl\bin;C:\Program Files\MiKTeX\miktex\bin\x64\;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin\;C:\Program Files\TortoiseGit\bin;C:\Program Files\nodejs\;C:\Users\glenn\AppData\Local\Microsoft\WindowsApps;C:\Program Files\OpenOCD\bin;C:\Users\glenn\AppData\Local\GitHubDesktop\bin;C:\Program Files\qemu;C:\Users\glenn;C:\msys64\usr\bin;C:\Users\glenn\AppData\Roaming\npm;C:\Users\glenn\AppData\Roaming\xPacks\@xpack-dev-tools\openocd\0.12.0-4.1\.content\bin\;C:\Oolite;
```

Somehow when VSCode opens it refreshes the paths on existing PowerShell terminals, but not existing Zephyr IDE terminals. Opening subsequent terminals have the correct path.

closed 2024-12-12 by rijesha