mylonics/zephyr-ide #86: Multiple identical entries added to Zephyr IDE Terminal Path

glenn-andrews, 2024-11-28:
```
C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Users\glenn\.zephyr_ide\.venv\Scripts;\;C:\Users\glenn\.zephyr_ide\.venv\bin;\;C:\Program Files (x86)\STMicroelectronics\stlink_server;C:\ST\STM32CubeCLT_1.16.0;C:\ST\STM32CubeCLT_1.16.0\CMake\bin;C:\ST\STM32CubeCLT_1.16.0\Ninja\bin;C:\ST\STM32CubeCLT_1.16.0\GNU-tools-for-STM32\bin;C:\ST\STM32CubeCLT_1.16.0\STLink-gdb-server\bin;C:\ST\STM32CubeCLT_1.16.0\STM32CubeProgrammer\bin;C:\Program Files\ImageMagick-7.1.1-Q16-HDRI;C:\Program Files\Eclipse Adoptium\jdk-21.0.3.9-hotspot\bin;C:\Python311\Scripts\;C:\Python311\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Microsoft VS Code\bin;C:\ProgramData\chocolatey\bin;C:\Program Files\CMake\bin;C:\Program Files\Git\cmd;C:\Program Files\LLVM\bin;C:\Program Files\doxygen\bin;C:\Strawberry\c\bin;C:\Strawberry\perl\site\bin;C:\Strawberry\perl\bin;C:\Program Files\MiKTeX\miktex\bin\x64\;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin\;C:\Program Files\TortoiseGit\bin;C:\Users\glenn\AppData\Local\Microsoft\WindowsApps;C:\Program Files\OpenOCD\bin;C:\Users\glenn\AppData\Local\GitHubDesktop\bin;C:\Program Files\qemu;C:\Users\glenn;
```
Do we need 66 copies of `C:\Users\glenn\.zephyr_ide\.venv\bin` added to the path?

glenn-andrews, 2024-11-28:
This seems to be called by using the Extension Setup and "Reset Workspace Settings" multiple times as I was experimenting.

rijesha, 2024-11-28:
This has been fixed in the prerelease 1.4.1. It is now available in release v1.5.0.

closed 2024-11-28 by rijesha