mylonics/zephyr-ide #88: Feature Request: Twister testing

glenn-andrews, 2024-11-30:
Add the ability to run Twister commands from the UI.

Twister is Zephyr's testing framework and can be invoked using the `west twister` command.

Testing can be performed in simulation using QEMU or native_sim by defining the platform (in this case QEMU) and runing all test cases in the current directory:
```
west twister -p qemu_x86 -T .
```

Hardware in the loop testing can also be performed:
```
west twister --device-testing --device-serial COM1 --device-serial-baud 115200 -p disco_l475_iot1 -T .
```
In this case both platform, serial port and baud rate need be specified.

You can specify individual test cases with the `-s` command, but that would require parsing the `testcase.yaml` or `sample.yaml` file in the project directory to list them, or providing a text box to enter manually.

Build files and test results are placed in the `twister_out` directory. Subsequent tests rename the previous directory to `twister_out.1` etc. A button to clean up all `twister_out*` directories would be appreciated.

rijesha, 2024-11-30:
Can you make a mockup of what the GUI should look like?
Should the elements be in a side panel and saved persistently in zephyr-ide.json or just run once with quickpicks from a command?

glenn-andrews, 2024-11-30:
[image]

I'd say the board and command-line arguments should be persistent. 

If you want to be fancy, look for a `-T` in the command-line, and if not present, affix `-T .` to run all tests in the application directory.

glenn-andrews, 2024-11-30:
> Build files and test results are placed in the `twister_out` directory. Subsequent tests rename the previous directory to `twister_out.1` etc. A button to clean up all `twister_out*` directories would be appreciated.

my mistake. It's `twister-out` with a hyphen, not an underscore.

rijesha, 2025-06-22:
Twister testing has finally been added in the latest prerelease.

@glenn-andrews please let me know if you are still interested and if you can test this feature.

glenn-andrews, 2025-06-22:
Definitely interested, but a bit overwhelmed right now. Should be able to look into this next month.

rijesha, 2025-06-23:
Thanks. I will leave this issue open for now then until you have a chance to look at it.

rijesha, 2025-08-01:
Hi Glen. Have you had a chance to look at it. I will close this issue and we can reopen it if you run into any problems.

closed 2025-08-01 by rijesha

glenn-andrews, 2025-08-01:
> Hi Glen. Have you had a chance to look at it. I will close this issue and we can reopen it if you run into any problems.

Sorry, work has me slammed. Closing it sounds good.