mylonics/zephyr-ide #90: IDE Unable to load custom board definition

StephenKercherPP, 2024-12-09:
We have implemented a custom board definition for zephyr.  It compiles and works perfectly with the West build system.
Importing that project into the Zephyr IDE extensions works perfectly.
However, when i try to specify the 'build', the extension asks for the Boards directory.  I specify my own directory (since the board definition is not in the zephyr/boards directory).
The extension errors out with the following message:

```
Traceback (most recent call last):
  File "c:\Users\this_user\.vscode\extensions\mylonics.zephyr-ide-1.5.2\scripts\get_board_list.py", line 51, in <module>
    get_boards(args)
  File "c:\Users\this_user\.vscode\extensions\mylonics.zephyr-ide-1.5.2\scripts\get_board_list.py", line 36, in get_boards
    log.inf(args.format.format(name=board.name, arch='', dir=board.dir, hwm=board.hwm,
                                    ^^^^^^^^^^
AttributeError: 'str' object has no attribute 'name'
```

This leads me to believe that either the extension does not support custom board definitions OR does not support where my custom board definition is located.
zephyr_boards:  <app_dir>/zephyr/boards
custom_boards: <app_dir>/app/boards

I tried modifying the python files in the extension to debug where the issue is happening but I have been unable to do so.

rijesha, 2024-12-09:
There is currently a bug where custom board directories is not working with global/external folder installs. Does the custom board directory work for you with a workspace/local install?

StephenKercherPP, 2024-12-09:
I am using a Workspace Install, i believe.
[image]

olk-jbozzo, 2024-12-09:
I am having the same issue. I am using a Workspace install.

StephenKercherPP, 2024-12-10:
For reference, i am on windows;
My device is a STM32F756BG.  The closest zephyr-defined board is a Nucleo_f756ZG.
It also appears that the IDE is correctly finding the board (using the find_boards python function) but is unable to parse the information there (since the foreach loop actually loops)

rijesha, 2024-12-12:
I have released a fix in v1.5.3, Can you let me know if the issue is resolved once that goes live.

closed 2024-12-12 by rijesha