mylonics/zephyr-ide #91: Exception creating a new custom board

olk-jbozzo, 2024-12-09:
I am working on a project that compiles and runs fine on a Zephyr-defined board.

Now I want to flash it on my custom board and I have issues creating it. I've been following this tutorial [https://www.youtube.com/watch?v=TXcTzyswBMQ](https://www.youtube.com/watch?v=TXcTzyswBMQ). But get this error when selecting a board folder different from the zephyr one (When adding a build configuration):

```
Traceback (most recent call last):
  File "/home/jbozzo/.vscode/extensions/mylonics.zephyr-ide-1.5.2/scripts/get_board_list.py", line 51, in <module>
    get_boards(args)
  File "/home/jbozzo/.vscode/extensions/mylonics.zephyr-ide-1.5.2/scripts/get_board_list.py", line 36, in get_boards
    log.inf(args.format.format(name=board.name, arch='', dir=board.dir, hwm=board.hwm,
                                    ^^^^^^^^^^
AttributeError: 'str' object has no attribute 'name'
```

Steps to reproduce
---

0. **Set up the python environment:**
   - With [pyenv](https://github.com/pyenv/pyenv) install Python 3.12.7.
   - With pyenv create a new virtual enviroment.
   - Activate environment.
   - Run `pip install west`.
   - Open vscode in the environment.
1. **Create an empty folder (name: example)**
2. **In Zephyr Ide Extension, initialize the workspace:**
    - Extension Setup  ->
    - Initialize Workspace 
        - Full Zephyr
        - Default version
        - No aditional arguments
        - Select every toolchain
3. **Create a new project based on LoRaWAN class A**
    - Zephyr IDE: Create project from template
    - Select class_a (subsys/lorawan/class_a)
    - Use class_a as folder name.
4. **Create a "boards/atmel/sam0" folder nested on the "example" folder**
5. **Derive a new board from samr34 xplained pro.**
    - Copy `external/zephyr/boards/atmel/sam0/samr34_xpro` into `boards/atmel/sam0/samr34_xpro`
    - Rename `boards/atmel/sam0/samr34_xpro` to `boards/atmel/sam0/samr34_xpro2`
    - Rename `boards/atmel/sam0/samr34_xpro/Kconfig.samr34_xpro` to `boards/atmel/sam0/samr34_xpro2/Kconfig.samr34_xpro2`
    -  Rename `boards/atmel/sam0/samr34_xpro/samr34_xpro_defconfig` to `boards/atmel/sam0/samr34_xpro2/samr34_xpro2_defconfig`
    -  Rename `boards/atmel/sam0/samr34_xpro/samr34_xpro.yaml` to `boards/atmel/sam0/samr34_xpro2/samr34_xpro2.yaml`
    -  Rename `boards/atmel/sam0/samr34_xpro/samr34_xpro.dts` to `boards/atmel/sam0/samr34_xpro2/samr34_xpro2.dts`
    -  Rename `boards/atmel/sam0/samr34_xpro/samr34_xpro-pinctrl.dtsi` to `boards/atmel/sam0/samr34_xpro2/samr34_xpro2-pinctrl.dtsi`
    - Change `BOARD_SAMR34_XPRO` to `BOARD_SAMR34_XPRO2` in `Kconfig.samr34_xpro2`
    - Change `board.name: samr34_xpro` to `board.name: samr34_xpro2` in `Kconfig.samr34_xpro2`
    - Change `#include "samr34_xpro-pinctrl.dtsi" to `#include "samr34_xpro2-pinctrl.dtsi"` in `samr34_xpro2.dts`
    - Change `identifier: samr34_xpro` to `identifier: samr34_xpro2` in `samr34_xpro2.yaml`
6. **(Good case) Create a new Build with Zephyr board**
    - Projects ->
    - class_a
        - Add Build
        - Pick board directory: `.../external/zephyr/boards`
        - Pick board: samr34_xpro
        - Name for build: build/samr34_xpro
        - Select Build Optimization: Debug
        - Additional arguments:
        - Modify CMake arguments: <keep defaults>
    - Compiles, flashes and runs fine
7. **(Bad case) Create a new Build with custom board**
    - Projects ->
    - class_a
        - Add Build
        - Pick board directory: `.../example/boards`
        - Error notification
        - Open Output -> Zephyr IDE
        - Look at the error

Attachment
---
I've included my example/boards directory in case there is something obvious I am missing.

[boards.zip](https://github.com/user-attachments/files/18068984/boards.zip)

Environment
---
OS: Ubuntu 24.04.1 LTS
Zephyr extension: 1.5.2
Vscode: 1.93.1

olk-jbozzo, 2024-12-09:
Probably a duplicate of #90

rijesha, 2024-12-11:
Thanks for the detailed bug report. I will try to replicate the issue and issue a fix soon.

rijesha, 2024-12-12:
I have released a fix in v1.5.3, Can you let me know if the issue is resolved once that goes live.

wyswys95, 2024-12-12:
I upgrade to v1.5.3. now boards from 2 folders show after choose boards folder in workspace folder. But I can choose "my board" and add.

rijesha, 2024-12-12:
Ok. That is the new behaviour. I was maintaining some custom scripts to only list boards from a single board directory. The latest version of zephyr now has that feature, but it seems buggy from my testing. Sometime in the future I may revisit it to get only a list from the current directory displayed.

closed 2024-12-12 by rijesha

olk-jbozzo, 2024-12-12:
I confirm this was solved for me too.