mylonics/zephyr-ide #93: Workspace .vscode directory is not created automatically

rpiper, 2024-12-26:
I noticed that on a fresh workspace ,without a pre-existing .vscode directory, the .vscode directory is not automatically created, and thus the compile_commands.json and zephyr-ide.json files fail to be created or saved.

I am on Windows 10/11, and have not confirmed this problem exists on any other OS.

rijesha, 2024-12-26:
I am on windows 11 and this has never happened for me. Whenever I setup a new folder it automatically creates a .vscode folder when I choose an install type.
My concern is that the oneTimeWorkspaceSetup may not be called, because the folder name you are using was already once setup with the extension. 
I think there could be a bug but not 100% sure the PR will fix it.
Could you see if the bug is present with a folder with a brand new name you have not used before in that specific directory?

rpiper, 2024-12-27:
I've tried re-initializing the workspace several times and manually deleting the .venv, .west, and .vscode directories, and the .vscode directory is never recreated like the others. This is starting to make sense, since references to it only use `fs.writeFile()` which will not create the parent directory of the file being written, if it does not exist.

Since it's using fs-extra, changing to `fs.outputFile()` will create the parent directory, and I have confirmed will create the .vscode directory.

rijesha, 2025-01-06:
Thanks for the PR @rpiper.I think I will make a follow up commit to try to rework how oneTimeWorkspaceSetup is called.

PR #94 (2025-01-06): Make .vscode directory if it doesn't exist
Resolves #93
closed 2025-01-06 by rijesha