mylonics/zephyr-ide #96: Not working on Xubuntu 24.10 with zsh

424778940z, 2025-01-05:
fresh install of Xubuntu 24.10 with zsh

when selecting west init or west update, error will popup says 
`The terminal process "/usr/bin/zsh '-c', 'west init -l /home/aaa/Desktop/okpro-zephyr/west-manifest '" failed to launch (exit code: 127).`

looks like python environment is not activated properly while running west

rijesha, 2025-01-06:
Do you also need to set terminal.integrated.defaultProfile?

rijesha, 2025-01-06:
Also is zsh the default terminal in xubuntu 24.10? I have been trying to provide better support for zsh and I could spin up a xubuntu 24.20 image to do further testing.

424778940z, 2025-01-06:
> Do you mind sending your entire settings.json file?

sure

```json
{
    "http.noProxy": [
        "127.0.0.1"
    ],
    "http.proxy": "http://127.0.0.1:10808",
    "http.proxyStrictSSL": false,
    "terminal.external.linuxExec": "",
    "terminal.integrated.sendKeybindingsToShell": true,
    "terminal.integrated.scrollback": 99999,
    "terminal.integrated.automationProfile.linux": {
        "path": "bash"
    },
    "git.openRepositoryInParentFolders": "always",
    "cmake.showOptionsMovedNotification": false,
    "zephyr-ide.tools_directory": "/home/adam/zephyrproject",
}
```

asgoldenberg, 2025-01-09:
I notice a similar problem, but on Mac (Sonoma 14.6.1 + VS Code 1.96.2 + Zephyr IDE 1.6.2):

Upon pressing 'Initialize Workspace' in the VS Code extension, all 4 steps work great until inside the 'West Update' step (5/5). When we get the point of updating the python environment, bash calls pip which directs to the global environment and not our zephyr .venv, no matter what I change in the settings. The global python environment is controlled by Homebrew, so the plugin's script fails because it's not allowed to install python packages. Installing on CLI by hand works fine (i.e. starting the .venv then issuing the same command), but then the VSCode plugin is not aware of the successful 'West Update' and continues to block compiling until 'West Update' has been run. So, for me this issue is a plugin showstopper.

Terminal in VS Code:
`"/bin/bash '-l', '-c', 'pip install -r .../external/zephyr/scripts/requirements.txt'" terminated with exit code: 1`

I'll try a few workarounds, or go back to Linux....

asgoldenberg, 2025-01-09:
Regarding the above comment, a proven workaround is:

1. Install the python dependencies manually, like:

`(.venv) prj $ pip install -r external/zephyr/scripts/requirements.txt`

2. Use the command pallette command:

`Zephyr IDE: Mark West as Ready`

rijesha, 2025-03-26:
@424778940z So I tried running with zsh and pop_os 22.04 and it worked correctly for me. 

I installed zsh with these commands. 

```
# zsh
sudo apt install -y zsh
sudo usermod --shell /bin/zsh root
sudo usermod --shell /bin/zsh $USER
```

I was thinking this was a zsh alone related issue, but now I am wondering if this is also related to xubuntu or grml-core as well.

When it fails, if you open a Zephyr IDE Terminal and run the command ```echo $PATH```, do you see the .venv folder in there?

rijesha, 2025-03-27:
@asgoldenberg @424778940z Can you try prerelease version 1.6.3 and see if the issue is fixed?

424778940z, 2025-03-28:
> Can you try prerelease version 1.6.3 and see if the issue is fixed?

Tested on latest version of vscode and 1.6.3 pre-release plugin, still giving the same error

```
 *  Executing task: west update 

zsh:1: command not found: west

 *  The terminal process "/usr/bin/zsh '-c', 'west update'" failed to launch (exit code: 127). 
 *  Terminal will be reused by tasks, press any key to close it.
```

424778940z, 2025-03-28:
> but now I am wondering if this is also related to xubuntu or grml-core as well.

hummm after further testing, the issue is indeed caused by `grml-core` config!

If I reset it back to default config, it works fine

```shell
sudo mv /etc/zsh /etc/zsh.grem
sudo mv /etc/zsh.bak /etc/zsh
mv ~/.zshrc ~/.zshrc.grem
touch ~/.zshrc
```

closed 2025-03-28 by 424778940z

rijesha, 2025-06-23:
@424778940z I may have a fix that could work even with grml-core installed. If you would like to give it a try you can with prerelease 1.7.7.
It is a macOS only fix that I currently have enabled for all platforms. If it works for you I can look into making it a configuration flag.