mylonics/zephyr-ide #101: Workspace setup failed on macOS Sonoma Apple Silicon, but completed after reopening VSCode

ekawahyu, 2025-02-28:
Workspace setup is kind of hit and miss on macOS Sonoma Apple Silicon. But eventually I can make it work after reinitializing it again. Please see the following steps to reproduce it:

1. Open VSCode and create an empty folder.
2. Go to Zephyr IDE and select Workspace.
3. Initialize Workspace, select anything really, in my case I set it up for STM32, Zephyr 3.7.0.
4. At this point, workspace setup will fail with zsh saying that west command is not found:

```
 *  Executing task: west init -l /Users/macos/zephyrstm32/west-manifest  

zsh:1: command not found: west

 *  The terminal process "/bin/zsh '-l', '-c', 'west init -l /Users/macos/zephyrstm32/west-manifest '" failed to launch (exit code: 127). 
 *  Terminal will be reused by tasks, press any key to close it. 
```

5. Close VSCode and then reopen the workspace.
6. Go to Zephyr IDE and Initialize Workspace again.
7. Select whatever you chose earlier, but this time there will be a pop up saying that Python Env already exists: choose Use Existing
8. Workspace setup is successful!

I am afraid I don't know how to debug this, but feel free to ask me anything. Thank you for this awesome plugin!

ekawahyu, 2025-02-28:
It could be related to https://github.com/mylonics/zephyr-ide/issues/96

ekawahyu, 2025-02-28:
A little bit of digging, this is what I see with the selected Python interpreter, before .venv is created:

[image]

This is after reopening the VSCode:

[image]

So clearly, the venv was not taken during the first setup.

ekawahyu, 2025-02-28:
Ah, I think to workaround it, when zsh says that west command is not found, you can manually select the Python Interpreter and then continue where it was left off. Hopefully this information helps you to fix this issue.

rijesha, 2025-03-03:
@ekawahyu Thanks for all the info. This is something that I have been trying to fix for a while now so the additional information is very useful

ekawahyu, 2025-03-04:
Is there debug log that I can turn on with VS Code extension? I am not familiar with writing extension, wondering if there is a way to debug this in the VSCode?

rijesha, 2025-03-26:
@ekawahyu  If you are interested in doing some deeper debugging/development on this issue that would be great. I have no way of testing with a Mac OS on Apple Silicon.

To start off you can setup the extension for development by following developer-guide.md.

So the main problem is that it seems like sometimes the python environment is not being detected.

In utils.ts there is a command called getShellEnvironment. This command returns the environment variables used when a command is run.

The "PATH" and "VIRTUAL_ENV" variable are the most relevant to this issue. They should be updated to the virtual environment.

Line 612 of setup.ts is where the error is occurring. So for some reason maybe "PATH" and/or "VIRTUAL_ENV" is not set up properly. You could manually hardcode a console.log of the return of getShellEnvironment right before line 612 or do a console.log of envPath in executeTaskHelper in utils.ts.

This may show whether or not it is being set before the function is being called.

The other thing that may be happening is that the variables are being set, but the wrong python executable is still being selected. This may be due to the order that the PATH variable is being set in line 117 of utils.ts.

Let me know if you have the time/desire to try to debug this further.

rijesha, 2025-03-26:
Here is some relevant issues and readings:
https://github.com/microsoft/vscode-python/issues/22743
https://github.com/microsoft/vscode-python/issues/22901

https://github.com/microsoft/vscode/issues/205102

rijesha, 2025-03-27:
@ekawahyu Can you try prerelease version 1.6.3 and see if the issue is fixed.

rijesha, 2025-06-15:
More related issues:
https://github.com/microsoft/vscode/issues/188235
https://github.com/microsoft/vscode/issues/179476
https://github.com/microsoft/vscode/issues/187955