mylonics/zephyr-ide #102: [Improvement] Configure Intellisense

JoseAmador95, 2025-03-18:
I love the extension, and I think the one big feature that is missing for me is to configure Intellisense. 

Could this extension be an Intellisense provider or just add a configuration in the workspace settings with the path to the active project’s compile_commands.json?

rijesha, 2025-03-19:
Hi. This extension should work with intellisense. The compile_commands.json for all the projects are compiled into a single file that is then pointed to in workspace settings. In .vscode, do you see a compile_commands.json? 
And in settings.json do you see the following line:
  "C_Cpp.default.compileCommands": "${workspaceFolder}\\.vscode\\compile_commands.json",

JoseAmador95, 2025-03-21:
This works! Thanks.

closed 2025-03-21 by JoseAmador95

JoseAmador95, 2025-03-26:
Actually, I had to add this config in my workspace.

```json
"C_Cpp.intelliSenseEngine": "default"
```

Otherwise the C/C++ extension output this.

```
intelliSenseEngine is disabled
```

I don't remember whether I set the path to the compile database or the extension did it.