mylonics/zephyr-ide #104: esptool not found

eiten, 2025-03-28:
Hello

fresh install on Win 10, latest VS code. ESP32C6. Compile works, but upon flashing, there is an error:

```
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner esp32
-- runners.esp32: reset after flashing requested
FATAL ERROR: required program C:/Users/iteedi/esp/external/modules/hal/espressif\tools\esptool_py\esptool.py not found; install it or add its location to PATH
```

Said file exists... did a "pip install esptool" in the venv but that did not help.

Regards, Edi

rijesha, 2025-03-28:
It seems to work just fine for me with powershell and a workspace install.

Are you using zephyr-ide 1.6.2 or 1.6.3? What version of zephyr are you using? Are you doing a global or workspace install?

rnicolas, 2025-04-17:
I also faced the same problem, there is a bug in zephyr itself. I already opened a [PR](https://github.com/zephyrproject-rtos/zephyr/pull/88605) to solve it. In the meantime you can take a look at this [workaround](https://github.com/zephyrproject-rtos/zephyr/discussions/88424#discussioncomment-12830475)

rnicolas, 2025-04-17:
But I'm wondering why in your setup @rijesha it works. I'm also using powershell and workspace install (the installation method mentioned by the zephyrproject team also gives me the same problem).

rijesha, 2025-04-17:
I'm not sure. One thought I had was perhaps I had run ```west blobs fetch hal_espressif``` and there is a fix in there for it?

rnicolas, 2025-04-19:
I also run it, and my problem still persists, I'm more inclined the problem comes the way python is installed, in my case I used winget, did you used winget or the python.org installation executables?

eiten, 2025-04-19:
Thanks! That solved my problem

rijesha, 2025-04-20:
I installed python so long ago. I think it was through the official installer. https://www.python.org/downloads/windows/

rnicolas, 2025-04-22:
@rijesha I've just tried with the official installer, problems are gone. It seems winget option misses some steps, but for what matters at zephyr-ide, I think you can just close this issue.

rijesha, 2025-04-22:
ok thanks.

closed 2025-04-22 by rijesha