mylonics/zephyr-ide #105: Project folders with periods in the name are not imported properly.

Matt-Armstrong-Faro, 2025-04-01:
When importing an existing project using the "Zephyr IDE: Add Project" command the project name is parsed incorrectly if the selected folder contains a period character. eg. MyProduct.Module1. This will cause the project to simply be named "MyProduct"... if you later try to add, say, MyProduct.Module2 then the add will fail since MyProduct already exists.

 - OS: Windows 11
 - vscode Version:  1.98.2
 - Zephyr IDE Version:  1.6.2
 - Zephyr Version:  4.0 
 - Zephyr SDK Version: 0.16.8 
 - Terminal: powershell

rijesha, 2025-04-02:
Thanks for the issue and PR. I will try to have a look and review this weekend.

rijesha, 2025-04-14:
Thanks for the PR and Issue. Your PR has been merged in. It will be available in the prerelease stream today.

closed 2025-04-14 by rijesha